chipsec/chipsec #958: setup.py build doesn't build extension (module driver anymore)

yaperez-anssi, 2020-07-10:
Hi,
I'm not sure if it's a real issue or not, so feel free to close as invalid if needed.

As part of the [chipsec-check](https://github.com/ANSSI-FR/chipsec-check) [build script](https://github.com/ANSSI-FR/chipsec-check/blob/master/tools/build-chipsec.sh), we are currently using `python3 setup.py install` has the only command for `build`, `build_ext` and `install`. Starting with `66f93f1b3d179f09e51d3d3a04707b4e03abad31` this is no longer possible, because `build` won't actually build the kernel modules anymore, and one needs to explicitely call `build_ext` (`install` will still call `build` if needed).

frinzell, 2021-05-03:
Hi, is this still a problem?  Hopefully, it has been fixed.  Thanks!

yaperez-anssi, 2021-05-06:
> Hi, is this still a problem? Hopefully, it has been fixed. Thanks!

I just tried a build with calling only `build install` (and not `build_ext`) and the extension is not built. So no the issue is not fixed, but then again maybe it's intended to be that way. For now I've changed our [build script](https://github.com/ANSSI-FR/chipsec-check/blob/master/tools/build-chipsec.sh) to call `build build_ext install` and it workarounds the issue.

npmitche, 2024-09-13:
Yes, this is working as intended. Please use build_ext.

closed 2024-09-13 by npmitche