chipsec/chipsec #962: Get an error install chipsec in ubuntu

wangzitiansky, 2020-07-22:
I install this but i get an erro
first i 
```bash
sudo apt-get install build-essential python-dev python-setuptools python gcc \ linux-headers-$(uname -r) nasm
```
then
```bash
sudo python setup.py install
```
then 
```bash
sudo python setup.py build_ext -i
```
when i do this
```bash
sudo python chipsec_util.py spi dump rom.bin
```
at last I get a error
```bash
################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version : 1.5.2
[CHIPSEC] OS      : Linux 5.4.0-42-generic #46-Ubuntu SMP Fri Jul 10 00:24:02 UTC 2020 x86_64
[CHIPSEC] Python  : 2.7.18rc1 (64-bit)

****** Chipsec Linux Kernel module is licensed under GPL 2.0
[CHIPSEC] API mode: using CHIPSEC kernel module API
ERROR: Chipset requires a supported PCH to be loaded: VID = 0xFFFF, DID = 0xFFFF, RID = 0xFF
WARNING: *******************************************************************
WARNING: * Unknown platform!
WARNING: * Platform dependent functionality will likely be incorrect
WARNING: * Error Message: "Chipset requires a supported PCH to be loaded: VID = 0xFFFF, DID = 0xFFFF, RID = 0xFF"
WARNING: *******************************************************************
ERROR: To run anyways please use -i command-line option

```

wangzitiansky, 2020-07-27:
I run it in a virtual machine, can the programmers run in a VM?

BrentHoltsclaw, 2020-07-29:
Yes chipsec can run within a VM, but what is the intended functionality that you are trying to get out of the tool.

wangzitiansky, 2020-07-29:
I want to use the command  
sudo python chipsec_util.py spi dump rom.bin
to get the bios into the rom.bin

frinzell, 2021-02-22:
No response, assume working.  Please reopen if needed.

closed 2021-02-22 by frinzell