chipsec/chipsec #974: run ：sudo chipsec_main;; report error :ERROR: Message: "Cannot find chipsec.ko module"

RockySYR, 2020-07-28:
run ：sudo chipsec_main;; report error :ERROR: Message: "Cannot find chipsec.ko module"
i need some help ,thx

npmitche, 2020-07-28:
You must build the driver. Please follow the steps outlined in the wiki here: https://github.com/chipsec/chipsec/wiki/Installing-CHIPSEC-in-Linux

RockySYR, 2020-07-28:
I have run these two commands ，apt-get install build-essential python-dev python gcc linux-headers-$(uname -r) nasm  AND python setup.py build_ext -i
No errors are reported, and I can run successfully # python chipsec_util.py spi dump rom.bin

RockySYR, 2020-07-28:
> You must build the driver. Please follow the steps outlined in the wiki here: https://github.com/chipsec/chipsec/wiki/Installing-CHIPSEC-in-Linux

i saw /divers/linux/readme.md ,
To build:
	make

To load kernel module:
	sudo insmod ./chipsec.ko
but But the same error occurs

BrentHoltsclaw, 2020-07-29:
@RockySYR what is the kernel version?  Chipsec currently does not support above version 5.7

RockySYR, 2020-07-29:
> kernel version

@BrentHoltsclaw ##Linux version 5.4.0-42-generic (buildd@lgw01-amd64-038) (gcc version 9.3.0 (Ubuntu 9.3.0-10ubuntu2)) #46-Ubuntu SMP Fri Jul 10 00:24:02 UTC 2020

closed 2020-08-03 by RockySYR