chipsec/chipsec #979: Unable to execute command ‘decode’

RockySYR, 2020-07-30:
fc@fc-virtual-machine:/tmp/chipsec-master$ sudo python chipsec_util.py decode lenovo_thinkpad_x1_carbon_6.bin
[sudo] fc 的密码： 

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version : 1.5.2
[CHIPSEC] OS      : Linux 5.4.0-42-generic #46-Ubuntu SMP Fri Jul 10 00:24:02 UTC 2020 x86_64
[CHIPSEC] Python  : 2.7.18rc1 (64-bit)

usage: 
    >>> chipsec_util decode <rom> [fw_type]

    For a list of fw types run:

    >>> chipsec_util decode types

    Examples:

    >>> chipsec_util decode spi.bin vss
    
chipsec_util.py: error: invalid choice: 'lenovo_thinkpad_x1_carbon_6.bin' (choose from 'types', 'decode')

BrentHoltsclaw, 2020-07-30:
try `chipsec_util uefi decode lenovo_thinkpad_x1_carbon_6.bin`

RockySYR, 2020-08-03:
> fc@fc-virtual-machine:/tmp/chipsec-master$ sudo python chipsec_util.py decode lenovo_thinkpad_x1_carbon_6.bin
> [sudo] fc 的密码：
> 
> ################################################################
> 
> ## CHIPSEC: Platform Hardware Security Assessment Framework
> ################################################################
> [CHIPSEC] Version : 1.5.2
> [CHIPSEC] OS : Linux 5.4.0-42-generic #46-Ubuntu SMP Fri Jul 10 00:24:02 UTC 2020 x86_64
> [CHIPSEC] Python : 2.7.18rc1 (64-bit)
> 
> usage:
> >>> chipsec_util decode [fw_type]
> 
> ```
> For a list of fw types run:
> 
> >>> chipsec_util decode types
> 
> Examples:
> 
> >>> chipsec_util decode spi.bin vss
> ```
> 
> chipsec_util.py: error: invalid choice: 'lenovo_thinkpad_x1_carbon_6.bin' (choose from 'types', 'decode')

closed 2020-08-03 by RockySYR