chipsec/chipsec #983: ERROR: Non-ASCII character '\xc3' in file

numbfx, 2020-08-12:
Hello, I'm getting the following error when i try to run different commands, here is an example:

`root@numbfx-MacBookPro:/home/numbfx/Downloads/chipsec# python chipsec_util.py msr

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version : 1.5.3
[CHIPSEC] OS      : Linux 5.4.0-42-generic #46-Ubuntu SMP Fri Jul 10 00:24:02 UTC 2020 x86_64
[CHIPSEC] Python  : 2.7.18rc1 (64-bit)

[CHIPSEC] API mode: using OS native API (not using CHIPSEC kernel module)
ERROR: Non-ASCII character '\xc3' in file /home/numbfx/Downloads/chipsec/chipsec/helper/linux/cpuid.py on line 7, but no encoding declared; see http://python.org/dev/peps/pep-0263/ for details (cpuid.py, line 7)`

I proceeded to add `# - *- coding: utf- 8 - *- ` to the top of the file mentioned above, then got this error:

`root@numbfx-MacBookPro:/home/numbfx/Downloads/chipsec# python chipsec_util.py msr

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version : 1.5.3
[CHIPSEC] OS      : Linux 5.4.0-42-generic #46-Ubuntu SMP Fri Jul 10 00:24:02 UTC 2020 x86_64
[CHIPSEC] Python  : 2.7.18rc1 (64-bit)

[CHIPSEC] API mode: using OS native API (not using CHIPSEC kernel module)
Segmentation fault (core dumped)
root@numbfx-MacBookPro:/home/numbfx/Downloads/chipsec#`

what's going on?

marcomafcorp, 2020-08-12:
Hi numbfx,

Sorry for the inconvenience with this, I've replicated the issue and here is the fix.

On the first line of /chipsec/chipsec/helper/linux/cpuid.py please add
# coding=utf-8 
or 
# - *- coding: utf- 8 - *- 

This should fix your issue because I tested both and it fixed the issue on my end. I will wait for you to reply to this comment before closing this issue. 

[image]

marcomafcorp, 2020-08-13:
Hi Numbfx,
Sorry for your inconvience on this issue.

Currently python 3 is POR, I've tested the above issue on python3 and I've confirmed that it is working.

We have confirmed the issue that you are having while using python2 and we will have a pull request addressing the issue after thoroughly testing the change.

marcomafcorp, 2020-08-20:
@numbfx is it okay if we close this issue.

numbfx, 2020-08-23:
Yes thank you, the issue is resolved.

closed 2020-08-24 by npmitche