chipsec/chipsec #985: Management Engine (ME) module implementation

mahdishakeri88, 2020-08-15:
Hi dears

I tested the security features of a Gen. 8 HP server (HPE ProLiant DL360P) by using ChipSec 1.5.1 tool. However, many modules are not implemented such as Management Engine (ME). As you known, this module is important to security check, I want to implement this module (ME) anyway.
Can anyone introduce me a proper tool to test this module?

Tnx

frinzell, 2020-08-17:
Hi,
Can you please confirm which module you are referring to?
Each may have their own requirements to run; for instance, me_mfg_mode.py checks if the MEI1 device is enabled.

Thanks

mahdishakeri88, 2020-08-31:
Hi dear Frinzell

"chipsec.modules.common.me_mfg.mode" is my desired module that Cannot implement in my platform by ChipSec. Do you have any idea to check the status (enable/disable) of this module?
Tnx for your replying

frinzell, 2020-08-31:
**me_mfg_mode.py** checks to see if the `MEI1` device is enabled. Your platform may have this device disabled.

frinzell, 2020-09-04:
Hi, was this helpful?  Can we close this?
Thanks!

mahdishakeri88, 2020-09-28:
Hi, yes it was very helpful

frinzell, 2020-09-28:
Thanks!  

Closing.  Reopen if needed.

closed 2020-09-28 by frinzell