chipsec/chipsec #1003: Segfault when trying to whitelist file

0xfede7c8, 2020-09-09:
I'm getting a segfault when running:

```bash
python3 chipsec/chipsec_main.py --no_driver -i -n -m tools.uefi.whitelist -a generate whitelist.json IM144_0183_B00.zip
```

I know the image attached is a ZIP file and does not necessarilly has to work with the whitelisting module, but anyway, a segfault shouldn't happen.

`Python 3.8.2`

[IM144_0183_B00.zip](https://github.com/chipsec/chipsec/files/5195633/IM144_0183_B00.zip)

0xfede7c8, 2020-09-09:
I also happens with the file inside the ZIP

BrentHoltsclaw, 2020-09-09:
@0xfede7c8 what does the output look like when you run with the -d flag?  Also are you running as a root?

0xfede7c8, 2020-09-09:
Output with `-d` flag:

```bash
[*] Ignoring unsupported platform warning and continue execution
################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.5.4
[CHIPSEC] Arguments: --no_driver -i -n -d -m tools.uefi.whitelist -a generate ./whitelist.json images/Apple/IM144_0183_B00.zip

[helper] Linux Helper created
[helper] Linux Helper started/loaded
[CHIPSEC] API mode: using OS native API (not using CHIPSEC kernel module)
[*] Couldn't import chipsec.cfg.CFL
No module named 'chipsec.cfg.CFL'
[*] Configuration Files:
[*] - /home/xfede/Desktop/chipsec/tests/regression/tools/chipsecmaster/chipsec/cfg/8086/apl.xml
[*] - /home/xfede/Desktop/chipsec/tests/regression/tools/chipsecmaster/chipsec/cfg/8086/avn.xml
[*] - /home/xfede/Desktop/chipsec/tests/regression/tools/chipsecmaster/chipsec/cfg/8086/bdw.xml
[*] - /home/xfede/Desktop/chipsec/tests/regression/tools/chipsecmaster/chipsec/cfg/8086/bdx.xml
[*] - /home/xfede/Desktop/chipsec/tests/regression/tools/chipsecmaster/chipsec/cfg/8086/byt.xml
[*] - /home/xfede/Desktop/chipsec/tests/regression/tools/chipsecmaster/chipsec/cfg/8086/cfl.xml
[*] - /home/xfede/Desktop/chipsec/tests/regression/tools/chipsecmaster/chipsec/cfg/8086/cht.xml
[*] - /home/xfede/Desktop/chipsec/tests/regression/tools/chipsecmaster/chipsec/cfg/8086/cml.xml
[*] - /home/xfede/Desktop/chipsec/tests/regression/tools/chipsecmaster/chipsec/cfg/8086/common.xml
[*] - /home/xfede/Desktop/chipsec/tests/regression/tools/chipsecmaster/chipsec/cfg/8086/dnv.xml
[*] - /home/xfede/Desktop/chipsec/tests/regression/tools/chipsecmaster/chipsec/cfg/8086/glk.xml
[*] - /home/xfede/Desktop/chipsec/tests/regression/tools/chipsecmaster/chipsec/cfg/8086/hsw.xml
[*] - /home/xfede/Desktop/chipsec/tests/regression/tools/chipsecmaster/chipsec/cfg/8086/hsx.xml
[*] - /home/xfede/Desktop/chipsec/tests/regression/tools/chipsecmaster/chipsec/cfg/8086/icl.xml
[*] - /home/xfede/Desktop/chipsec/tests/regression/tools/chipsecmaster/chipsec/cfg/8086/iommu.xml
[*] - /home/xfede/Desktop/chipsec/tests/regression/tools/chipsecmaster/chipsec/cfg/8086/ivb.xml
[*] - /home/xfede/Desktop/chipsec/tests/regression/tools/chipsecmaster/chipsec/cfg/8086/ivt.xml
[*] - /home/xfede/Desktop/chipsec/tests/regression/tools/chipsecmaster/chipsec/cfg/8086/jkt.xml
[*] - /home/xfede/Desktop/chipsec/tests/regression/tools/chipsecmaster/chipsec/cfg/8086/kbl.xml
[*] - /home/xfede/Desktop/chipsec/tests/regression/tools/chipsecmaster/chipsec/cfg/8086/pch_1xx.xml
[*] - /home/xfede/Desktop/chipsec/tests/regression/tools/chipsecmaster/chipsec/cfg/8086/pch_2xx.xml
[*] - /home/xfede/Desktop/chipsec/tests/regression/tools/chipsecmaster/chipsec/cfg/8086/pch_3xx.xml
[*] - /home/xfede/Desktop/chipsec/tests/regression/tools/chipsecmaster/chipsec/cfg/8086/pch_3xxlp.xml
[*] - /home/xfede/Desktop/chipsec/tests/regression/tools/chipsecmaster/chipsec/cfg/8086/pch_3xxop.xml
[*] - /home/xfede/Desktop/chipsec/tests/regression/tools/chipsecmaster/chipsec/cfg/8086/pch_495.xml
[*] - /home/xfede/Desktop/chipsec/tests/regression/tools/chipsecmaster/chipsec/cfg/8086/pch_4xx.xml
[*] - /home/xfede/Desktop/chipsec/tests/regression/tools/chipsecmaster/chipsec/cfg/8086/pch_4xxh.xml
[*] - /home/xfede/Desktop/chipsec/tests/regression/tools/chipsecmaster/chipsec/cfg/8086/pch_4xxlp.xml
[*] - /home/xfede/Desktop/chipsec/tests/regression/tools/chipsecmaster/chipsec/cfg/8086/pch_c60x.xml
[*] - /home/xfede/Desktop/chipsec/tests/regression/tools/chipsecmaster/chipsec/cfg/8086/pch_c61x.xml
[*] - /home/xfede/Desktop/chipsec/tests/regression/tools/chipsecmaster/chipsec/cfg/8086/pch_c620.xml
[*] - /home/xfede/Desktop/chipsec/tests/regression/tools/chipsecmaster/chipsec/cfg/8086/qrk.xml
[*] - /home/xfede/Desktop/chipsec/tests/regression/tools/chipsecmaster/chipsec/cfg/8086/sfdp.xml
[*] - /home/xfede/Desktop/chipsec/tests/regression/tools/chipsecmaster/chipsec/cfg/8086/skl.xml
[*] - /home/xfede/Desktop/chipsec/tests/regression/tools/chipsecmaster/chipsec/cfg/8086/skx.xml
[*] - /home/xfede/Desktop/chipsec/tests/regression/tools/chipsecmaster/chipsec/cfg/8086/snb.xml
[*] - /home/xfede/Desktop/chipsec/tests/regression/tools/chipsecmaster/chipsec/cfg/8086/whl.xml
[*] - /home/xfede/Desktop/chipsec/tests/regression/tools/chipsecmaster/chipsec/cfg/template.xml
[*] Loading Configuration Files:
[*] looking for platform config in '/home/xfede/Desktop/chipsec/tests/regression/tools/chipsecmaster/chipsec/cfg/8086/common.xml'..
[*] loading common platform config from '/home/xfede/Desktop/chipsec/tests/regression/tools/chipsecmaster/chipsec/cfg/8086/common.xml'..
[*] loading integrated devices/controllers..
    + HOSTCTRL        : {'bus': '0', 'dev': '0x00', 'fun': '0', 'vid': '0x8086'}
    + PEG10           : {'bus': '0', 'dev': '0x01', 'fun': '0', 'vid': '0x8086'}
    + PEG11           : {'bus': '0', 'dev': '0x01', 'fun': '1', 'vid': '0x8086'}
    + PEG12           : {'bus': '0', 'dev': '0x01', 'fun': '2', 'vid': '0x8086'}
    + IGD             : {'bus': '0', 'dev': '0x02', 'fun': '0', 'vid': '0x8086'}
    + HDAUDIO         : {'bus': '0', 'dev': '0x03', 'fun': '0', 'vid': '0x8086'}
    + XHCI            : {'bus': '0', 'dev': '0x14', 'fun': '0', 'vid': '0x8086'}
[... cut ...]
    + SMILock         : {'register': 'GEN_PMCON_2', 'field': 'SMI_LOCK', 'desc': 'SMI Global Configuration Lock'}
[*] looking for platform config in '/home/xfede/Desktop/chipsec/tests/regression/tools/chipsecmaster/chipsec/cfg/8086/pch_3xxlp.xml'..
[*] looking for platform config in '/home/xfede/Desktop/chipsec/tests/regression/tools/chipsecmaster/chipsec/cfg/8086/pch_3xxop.xml'..
[*] Discovering Bus Configuration:
[*] loading device buses..
[1]    231000 segmentation fault (core dumped)  python3 ./tools/chipsecmaster/chipsec_main.py --no_driver -i -n -d -m  -a
```

BrentHoltsclaw, 2020-09-09:
The problem actually appears to be in the chipset initialization routines (`init_cfg_bus` function).  Running as root would most likely solve the problem.  Long term it would probably make sense to create a separate `chipsec_decompress.py`.  If running as root is not sufficient, let's look into adding an option to skip loading the configuration.

closed 2020-10-05 by 0xfede7c8

kerneis-anssi, 2021-12-17:
I've run into the same issue, but it wasn't solved by running as root (probably because my kernel is locked down so raw io port access is restricted). For the record, to debug a segfault in Python, set the environment variable PYTHONFAULTHANDLER=1. It will give you a nice stack trace showing exactly where the fault occurs. In that case, it was in `chipsec/chipsec/helper/linux/legacy_pci.py` (used because the `--no-driver` option is passed).