chipsec/chipsec #1005: Support Boot Hole

konradwilk, 2020-09-11:
Would it be possible for CHIPSEC to support testing for the Boot Hole exploit? Aka https://eclypsium.com/2020/07/29/theres-a-hole-in-the-boot/ ?

Thanks!

wrffrz, 2020-09-11:
Hi Konrad, nice to see you over here also.  :)

It's pretty easy to make new module tests which run as part of chipsec_main.py and there's a couple things we can do there to identify BootHole-related issues:

1) Just do the basic checks of whether the system has a revoked shim or bootloader that's signed by a revoked cert.  
2) Parse the dbx variable to determine if the updated revocation list has been applied yet.

That second one is more interesting from a CHIPSEC perspective and we currently support using the native API via /sys/ to list and get UEFI variable contents as well as ioctls in the chipsec.ko kernel driver.

In addition, we could also manually parse the dbx variable out of the nvram region from the SPI dump, but that's a little harder as there are a few different nvram formats that different vendors have used.

But overall, it definitely seems feasible to add support to CHIPSEC to identify systems affected by the BootHole vulnerabilities.

npmitche, 2021-02-27:
I haven't spent much time looking for it, but @wrffrz , do you have the hash or guid for the revoked cert(s)?

johnloucaides, 2021-02-28:
I believe the latest dbx (containing all the revoked certs) is published here: https://uefi.org/revocationlistfile

wrffrz, 2021-02-28:
Thanks @johnloucaides, yes, that's where the latest dbxupdate files are hosted.

The dbxupdate has been split into arch-specific updates to save variable space, so there are now x86, x64, and arm64 revocation files instead of just a single unified one with all of the hashes.  Most of those are x64, but the the total list of revoked items at this time is 243 file hashes and 3 certificates.

An easy way to extract the hashes and certs that have been revoked is to use https://github.com/rhboot/dbxtool, but I'll attach the list as of today to this comment.

[x64_hashes.txt](https://github.com/chipsec/chipsec/files/6057609/x64_hashes.txt)
[x86_hashes.txt](https://github.com/chipsec/chipsec/files/6057611/x86_hashes.txt)
[arm64_hashes.txt](https://github.com/chipsec/chipsec/files/6057612/arm64_hashes.txt)