chipsec/chipsec #1007: These files cannot be read？

RockySYR, 2020-09-16:
ERROR: Unable to open file 'sect01_khez' for read access
ERROR: Unable to open file 'sect01_uoyj' for read access
ERROR: Unable to open file 'sect01_tipy' for read access
ERROR: Unable to open file 'sect01_ppil' for read access
ERROR: Unable to open file 'sect01_jthc' for read access
ERROR: Unable to open file 'sect01_fynn' for read access
ERROR: Unable to open file 'sect01_gmhv' for read access
ERROR: Unable to open file 'sect01_tcru' for read access
ERROR: Unable to open file 'sect00_aibi' for read access
ERROR: Unable to open file 'sect00_khqh' for read access
ERROR: Unable to open file 'sect00_pmob' for read access
ERROR: Unable to open file 'sect00_vdso' for read access
ERROR: Unable to open file 'sect01_gxpe' for read access
ERROR: Unable to open file 'sect01_hlia' for read access
ERROR: Unable to open file 'sect01_qlud' for read access
ERROR: Unable to open file 'sect01_nemg' for read access
ERROR: Unable to open file 'sect01_abwi' for read access
ERROR: Unable to open file 'sect01_zgzk' for read access
ERROR: Unable to open file 'sect01_dnsj' for read access
ERROR: Unable to open file 'sect00_swkc' for read access
ERROR: Unable to open file 'sect01_tiah' for read access
ERROR: Unable to open file 'sect01_rctx' for read access
ERROR: Unable to open file 'sect01_jbdn' for read access
ERROR: Unable to open file 'sect01_vmja' for read access
ERROR: Unable to open file 'sect01_dsun' for read access
ERROR: Unable to open file 'sect01_vrgz' for read access
ERROR: Unable to open file 'sect01_cale' for read access
ERROR: Unable to open file 'sect01_ikld' for read access
ERROR: Unable to open file 'sect01_imby' for read access
ERROR: Unable to open file 'sect01_xczp' for read access
ERROR: Unable to open file 'sect01_tkww' for read access
ERROR: Unable to open file 'sect01_zmmf' for read access
ERROR: Unable to open file 'sect01_ixtl' for read access
ERROR: Unable to open file 'sect01_pzws' for read access
ERROR: Unable to open file 'sect01_rxpm' for read access
ERROR: Unable to open file 'sect02_vdao' for read access
ERROR: Unable to open file 'sect01_iyea' for read access
ERROR: Unable to open file 'sect01_tbzm' for read access
ERROR: Unable to open file 'sect01_kdgs' for read access
ERROR: Unable to open file 'sect01_ocjo' for read access
ERROR: Unable to open file 'sect01_pbut' for read access
ERROR: Unable to open file 'sect01_ppwj' for read access
ERROR: Unable to open file 'sect01_pdse' for read access
ERROR: Unable to open file 'sect01_gbbq' for read access
ERROR: Unable to open file 'sect01_kvnb' for read access @tweksteen @rountree @hughsie @mtnmts @WilliamLeara

frinzell, 2020-09-17:
Can you verify that you are using Python 3?

frinzell, 2020-10-12:
Following up, are you still seeing an issue?

frinzell, 2021-02-22:
No response, assuming working.  Please reopen if needed.

closed 2021-02-22 by frinzell