chipsec/chipsec #1023: Software has write access to GBe region in SPI flash

mobahi99, 2020-10-12:
Hi, I have analyzed a 4th generation processor into a HP EliteDesk 800 G1 desktop and I have got some errors and warnings suchs us, software has write access to GBe region in SPI flash. I have analyzed another computers with the same warning. Is it correct or it could be any error. On the other, I have got an S3 boot script error. So, has firmware been compromised?

Moreover, I have got the warning: Variables with attributes that differ from UEFI spec. How could I know which variables have this issue?

Last, I have got another warning in the chipsec.modules.common.cpu.spectre_v2. It's CPU supports mitigation (IBRS) but doesn't support enhanced IBRS. What its means?
Thank you for your help.

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.3.7
[CHIPSEC] Arguments: 
****** Chipsec Linux Kernel module is licensed under GPL 2.0
[CHIPSEC] API mode: using CHIPSEC kernel module API
[CHIPSEC] OS      : Linux 4.19.0-kali4-amd64 #1 SMP Debian 4.19.28-2kali1 (2019-03-18) x86_64
[CHIPSEC] Platform: Desktop 4th Generation Core Processor (Haswell CPU / Lynx Point PCH)
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 0C00
[CHIPSEC] PCH     : Default PCH
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 8C4E
 
[*] loading common modules from "/usr/local/lib/python2.7/dist-packages/chipsec-1.3.7-py2.7-linux-x86_64.egg/chipsec/modules/common" ..
[+] loaded chipsec.modules.common.spi_lock
[+] loaded chipsec.modules.common.spi_fdopss
[+] loaded chipsec.modules.common.spi_desc
[+] loaded chipsec.modules.common.spi_access
[+] loaded chipsec.modules.common.smrr
[+] loaded chipsec.modules.common.smm
[+] loaded chipsec.modules.common.sgx_check
[+] loaded chipsec.modules.common.rtclock
[+] loaded chipsec.modules.common.memlock
[+] loaded chipsec.modules.common.me_mfg_mode
[+] loaded chipsec.modules.common.ia32cfg
[+] loaded chipsec.modules.common.bios_wp
[+] loaded chipsec.modules.common.bios_ts
[+] loaded chipsec.modules.common.bios_smi
[+] loaded chipsec.modules.common.bios_kbrd_buffer
[+] loaded chipsec.modules.common.uefi.s3bootscript
[+] loaded chipsec.modules.common.uefi.access_uefispec
[+] loaded chipsec.modules.common.secureboot.variables
[+] loaded chipsec.modules.common.cpu.spectre_v2
[*] loading platform specific modules from "/usr/local/lib/python2.7/dist-packages/chipsec-1.3.7-py2.7-linux-x86_64.egg/chipsec/modules/hsw" ..
[*] loading modules from "/usr/local/lib/python2.7/dist-packages/chipsec-1.3.7-py2.7-linux-x86_64.egg/chipsec/modules" ..
[+] loaded chipsec.modules.smm_dma
[+] loaded chipsec.modules.remap
[+] loaded chipsec.modules.memconfig
[+] loaded chipsec.modules.debugenabled
[*] running loaded modules ..

[*] running module: chipsec.modules.common.spi_lock
[x][ =======================================================================
[x][ Module: SPI Flash Controller Configuration Locks
[x][ =======================================================================
[*] HSFS = 0xF008 << Hardware Sequencing Flash Status Register (SPIBAR + 0x4)
    [00] FDONE            = 0 << Flash Cycle Done 
    [01] FCERR            = 0 << Flash Cycle Error 
    [02] AEL              = 0 << Access Error Log 
    [03] BERASE           = 1 << Block/Sector Erase Size 
    [05] SCIP             = 0 << SPI cycle in progress 
    [13] FDOPSS           = 1 << Flash Descriptor Override Pin-Strap Status 
    [14] FDV              = 1 << Flash Descriptor Valid 
    [15] FLOCKDN          = 1 << Flash Configuration Lock-Down 
[+] SPI Flash Controller configuration is locked
[+] PASSED: SPI Flash Controller locked correctly.

[*] running module: chipsec.modules.common.spi_fdopss
[x][ =======================================================================
[... 622 lines cut ...]
[+] PASSED: chipsec.modules.common.smm
[+] PASSED: chipsec.modules.common.rtclock
[+] PASSED: chipsec.modules.common.memlock
[+] PASSED: chipsec.modules.common.me_mfg_mode
[+] PASSED: chipsec.modules.common.ia32cfg
[+] PASSED: chipsec.modules.common.bios_wp
[+] PASSED: chipsec.modules.common.bios_ts
[+] PASSED: chipsec.modules.common.bios_smi
[+] PASSED: chipsec.modules.common.uefi.access_uefispec
[+] PASSED: chipsec.modules.common.secureboot.variables
[+] PASSED: chipsec.modules.smm_dma
[+] PASSED: chipsec.modules.remap
[+] PASSED: chipsec.modules.memconfig
[+] PASSED: chipsec.modules.debugenabled
[CHIPSEC] Modules information     0:
[CHIPSEC] Modules failed          1:
[-] FAILED: chipsec.modules.common.uefi.s3bootscript
[CHIPSEC] Modules with warnings   3:
[!] WARNING: chipsec.modules.common.spi_access
[!] WARNING: chipsec.modules.common.bios_kbrd_buffer
[!] WARNING: chipsec.modules.common.cpu.spectre_v2
[CHIPSEC] Modules not implemented 0:
[CHIPSEC] Modules not applicable  1:
[*] NOT APPLICABLE: chipsec.modules.common.sgx_check
[CHIPSEC] *****************************************************************

npmitche, 2020-10-12:
Hi mobahi99,

> Software has write access to GBe region in SPI flash. I have analyzed another computers with the same warning. Is it correct or it could be any error. 

As chipsec is able to read and identify the fields correctly, this looks correct. 

> I have got an S3 boot script error. So, has firmware been compromised?

This test is not able to determine if your firmware has been compromised. It only verifies if your s3 boot script is vulnerable to modifications. More recent platforms hide the S3 boot script in an SMM (System Management Mode) lockbox. 

> Variables with attributes that differ from UEFI spec. How could I know which variables have this issue?

None have the issue. In later versions of chipsec, we only print that line if there are variables that do not match the UEFI spec. In this version you're running the variable name would be printed under that line. It is recommended to update to the latest version of chipsec (v1.5.5)

> Last, I have got another warning in the chipsec.modules.common.cpu.spectre_v2. It's CPU supports mitigation (IBRS) but doesn't support enhanced IBRS. What its means?

It is a hardware enhancement that provides better (more performant) protection against Spectre v2. If you do a web search for "enhanced IBRS" you can find more information.

mobahi99, 2020-10-13:
Thanks, Could you help me please with this other warning:

[x][ =======================================================================
[x][ Module: Check SGX feature support
[x][ =======================================================================
[*] Test if CPU has support for SGX
[+] SGX is supported on CPU

[*] SGX BIOS enablement check
[*] Verifying IA32_FEATURE_CONTROL MSR is configured
[!] WARNING: Intel SGX is not enabled in BIOS

[*] Verifying IA32_FEATURE_CONTROL MSR is locked
[+] IA32_Feature_Control locked

[*] Verifying if Protected Memory Range (PRMRR) is configured
[+] Protected Memory Range configuration is supported

[*] Verifying PRMRR Configuration on each core.
[+] PRMRR config is uniform across all CPUs
[*]  PRMRR config supports: 1M, 32M, 64M, 128M
[*]  PRMRR base address: 0x000000000000
[*]  Verifying PRMR memory type is valid
[*]  PRMRR memory type : 0x0
[-] FAILED: Unexpected PRMRR memory type (not WB)
[*]  PRMRR mask address: 0x000000000000
[*]  Verifying PRMR address are valid
[*]      PRMRR mask valid: 0x0
[-] FAILED: Mcheck marked PRMRR address as invalid
[*]  Verifying if PRMR mask register is locked
[*]      PRMRR mask lock: 0x0
[-] FAILED: PRMRR MASK register is not locked
[*]  PRMRR uncore base address: 0x000000000000
[*]  PRMRR uncore mask address: 0x000000000000
[*]  Verifying PRMR uncore address are valid
[*]      PRMRR uncore mask valid: 0x0
[-] FAILED: Mcheck marked uncore PRMRR address as invalid
[*]  Verifying if PRMR uncore mask register is locked
[*]      PRMRR uncore mask lock: 0x1
[+] PMRR uncore MASK register is locked

[*] Verifying if SGX is available to use
[!] WARNING: Intel SGX instructions disabled by firmware

and this other please:

[*] running module: chipsec.modules.common.rtclock
[x][ =======================================================================                                                                          
[x][ Module: Protected RTC memory locations                                
[x][ =======================================================================                                                                          
[*] RC = 0x00000004 << RTC Configuration (RCBA + 0x3400)
    [02] UE               = 1 << Upper 128 Byte Enable 
    [03] LL               = 0 << Lower 128 Byte Lock 
    [04] UL               = 0 << Upper 128 Byte Lock 
[-] Protected bytes (0x38-0x3F) in low 128-byte bank of RTC memory are not locked                                                                     
[-] Protected bytes (0x38-0x3F) in high 128-byte bank of RTC memory are not locked                                                                    
[!] WARNING: Protected locations in RTC memory are accessible (BIOS may not be using them)                                                            

Thank you.

npmitche, 2020-10-22:
Sorry for the delay.

> [!] WARNING: Intel SGX is not enabled in BIOS
> [!] WARNING: Intel SGX instructions disabled by firmware

These warning are the because SGX is not enabled. Assuming there's an option in your BIOS setup, if you turn SGX on, these warnings should go away.

> [!] WARNING: Protected locations in RTC memory are accessible (BIOS may not be using them)

This warning states that the memory locations generally used to store the RTC are not protected. Depending on the BIOS implementation, these locations might not be used, so it doesn't matter if that's the case. If they are used, it might be possible to change the values which could cause issues with things that require a valid time such as certificates.

frinzell, 2021-02-22:
Appears answered. Closing. Reopen if needed.

closed 2021-02-22 by frinzell