chipsec/chipsec #1024: error on decode " bytes_le is not a 16-char string"

RockySYR, 2020-10-14:
~$ python3 chipsec_util.py decode rom.bin

Traceback (most recent call last):
  File "chipsec_util.py", line 213, in <module>
    sys.exit( main() )
  File "chipsec_util.py", line 209, in main
    return chipsecUtil.main()
  File "chipsec_util.py", line 184, in main
    comm.run()
  File "/home/fc/chipsec5.5/chipsec/utilcmd/decode_cmd.py", line 126, in run
    self.decode_rom()
  File "/home/fc/chipsec5.5/chipsec/utilcmd/decode_cmd.py", line 116, in decode_rom
    decode_uefi_region(_uefi, pth, fname, self._fwtype)
  File "/home/fc/chipsec5.5/chipsec/hal/spi_uefi.py", line 480, in decode_uefi_region
    _uefi.parse_EFI_variables( nvram_fname, region_data, False, fwtype )
  File "/home/fc/chipsec5.5/chipsec/hal/uefi.py", line 363, in parse_EFI_variables
    decode_EFI_variables( efi_vars, nvram_pth )
  File "/home/fc/chipsec5.5/chipsec/hal/uefi.py", line 231, in decode_EFI_variables
    parse_efivar_file(var_fname, data, SECURE_BOOT_SIG_VAR)
  File "/home/fc/chipsec5.5/chipsec/hal/uefi_common.py", line 443, in parse_efivar_file
    parse_sb_db(var, var_path)
  File "/home/fc/chipsec5.5/chipsec/hal/uefi_common.py", line 314, in parse_sb_db
    owner = EFI_GUID_STR(owner0)
  File "/home/fc/chipsec5.5/chipsec/hal/uefi_common.py", line 192, in EFI_GUID_STR
    guid_str = UUID(bytes_le=guid)
  File "/usr/lib/python3.8/uuid.py", line 173, in __init__
    raise ValueError('bytes_le is not a 16-char string')
ValueError: bytes_le is not a 16-char string @tweksteen @rountree @hughsie @mtnmts @WilliamLeara

RockySYR, 2020-10-30:
it can‘t decode my rom file

RockySYR, 2020-10-30:
@ @rountree @tweksteen @mtnmts @hughsie @

BrentHoltsclaw, 2020-10-30:
@RockySYR please let us know if #1038 fixes the issue

PR #1038 (2020-11-19): Pass string not array to EFI_GUID_STR from parse_sb_db
Should fix #1024
closed 2020-11-19 by npmitche