chipsec/chipsec #1035: Problems building CHIPSEC windows driver

nickylimjj, 2020-10-23:
Hi, I followed the readme instructions but am met with the following error

```
Severity	Code	Description	Project	File	Line	Suppression State
Error	MSB3073	The command "mkdir ..\..\chipsec\helper\win\win7_amd64
copy C:\Users\USER\opt\chipsec-1.5.5\drivers\win7\x64\Debug\chipsec_hlpr.sys ..\..\chipsec\helper\win\win7_amd64
:VCEnd" exited with code 1.	chipsec_hlpr	C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets	149	

```

I am running Visual Studio 2019 and have installed WDK 10.0.19041. I have also installed the Spectre-mitigation libraries, as requested by the build script.

nickylimjj, 2020-10-23:
After investigating, I believe it is because the build script erroneously parses the username due to the space.

npmitche, 2021-02-27:
@nickylimjj  I was able to reproduce the issue and agree. Can you verify that #1109 fixes the issue for you?

frinzell, 2021-04-16:
Closing, fix merged.   Reopen if needed.  Thanks!

closed 2021-04-16 by frinzell