chipsec/chipsec #1039: chipsec fails to build driver on Pop!_OS (Ubuntu) 20.10 amd64; ~/chipsec/build/lib/drivers/linux/amd64/.cpu.o.cmd: No such file or directory

unilock, 2020-11-07:
I'm trying to build chipsec on Pop!_OS (Ubuntu) 20.10 amd64. When I run the build command indicated in the wiki, I run into a `make` error:
```
unilock@pop-os:~/chipsec$ python3 setup.py build_ext -i
running build_ext
building the linux driver
creating /home/unilock/chipsec/build
creating /home/unilock/chipsec/build/lib
creating /home/unilock/chipsec/build/lib/drivers
creating /home/unilock/chipsec/build/lib/drivers/linux
copying drivers/linux/README -> /home/unilock/chipsec/build/lib/drivers/linux
copying drivers/linux/AUTHORS -> /home/unilock/chipsec/build/lib/drivers/linux
copying drivers/linux/Makefile -> /home/unilock/chipsec/build/lib/drivers/linux
creating /home/unilock/chipsec/build/lib/drivers/linux/include
copying drivers/linux/include/chipsec.h -> /home/unilock/chipsec/build/lib/drivers/linux/include
copying drivers/linux/COPYING -> /home/unilock/chipsec/build/lib/drivers/linux
creating /home/unilock/chipsec/build/lib/drivers/linux/amd64
copying drivers/linux/amd64/cpu.asm -> /home/unilock/chipsec/build/lib/drivers/linux/amd64
creating /home/unilock/chipsec/build/lib/drivers/linux/i386
copying drivers/linux/i386/cpu.asm -> /home/unilock/chipsec/build/lib/drivers/linux/i386
copying drivers/linux/dkms.conf -> /home/unilock/chipsec/build/lib/drivers/linux
copying drivers/linux/chipsec_km.c -> /home/unilock/chipsec/build/lib/drivers/linux
copying drivers/linux/WARNING.txt -> /home/unilock/chipsec/build/lib/drivers/linux
/home/unilock/chipsec/build/lib/drivers/linux/amd64/.cpu.o.cmd: No such file or directory
make[2]: *** [scripts/Makefile.modpost:111: /home/unilock/chipsec/build/lib/drivers/linux/Module.symvers] Error 1
make[1]: *** [Makefile:1693: modules] Error 2
make: *** [Makefile:27: chipsec] Error 2
Traceback (most recent call last):
  File "setup.py", line 248, in <module>
    setup(
  File "/usr/lib/python3/dist-packages/setuptools/__init__.py", line 163, in setup
    return distutils.core.setup(**attrs)
  File "/usr/lib/python3.8/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/usr/lib/python3.8/distutils/dist.py", line 966, in run_commands
    self.run_command(cmd)
  File "/usr/lib/python3.8/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "setup.py", line 173, in run
    self._build_linux_driver()
  File "setup.py", line 81, in _build_linux_driver
    subprocess.check_output(["make", "-C", build_driver])
  File "/usr/lib/python3.8/subprocess.py", line 411, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.8/subprocess.py", line 512, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['make', '-C', '/home/unilock/chipsec/build/lib/drivers/linux']' returned non-zero exit status 2.
```

startergo, 2020-12-02:
Same in Ubuntu:
```
root@ubuntu:/home/g5/chipsec/drivers/linux# make
make -C /lib/modules/`uname -r`/build M=/home/g5/chipsec/drivers/linux clean
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-56-generic'
  CLEAN   /home/g5/chipsec/drivers/linux/Module.symvers
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-56-generic'
rm -f amd64/cpu.o
nasm -f elf64 -o amd64/cpu.o amd64/cpu.asm
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-56-generic'
  CC [M]  /home/g5/chipsec/drivers/linux/chipsec_km.o
  LD [M]  /home/g5/chipsec/drivers/linux/chipsec.o
  Building modules, stage 2.
  MODPOST 1 modules
WARNING: could not find /home/g5/chipsec/drivers/linux/amd64/.cpu.o.cmd for /home/g5/chipsec/drivers/linux/amd64/cpu.o
  CC [M]  /home/g5/chipsec/drivers/linux/chipsec.mod.o
  LD [M]  /home/g5/chipsec/drivers/linux/chipsec.ko
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-56-generic'

```

tinylabs, 2021-01-06:
Trying to build for Pop!_OS 20.10 and this appears to be an error. Even make -k will not complete. Anyone get past this?
elliot@pop-os:~/projects/me_cleaner/chipsec/drivers/linux$ make -k
make -C /lib/modules/`uname -r`/build M=/home/elliot/projects/me_cleaner/chipsec/drivers/linux clean
make[1]: Entering directory '/usr/src/linux-headers-5.8.0-7630-generic'
make[1]: Leaving directory '/usr/src/linux-headers-5.8.0-7630-generic'
rm -f amd64/cpu.o
nasm -f elf64 -o amd64/cpu.o amd64/cpu.asm
make[1]: Entering directory '/usr/src/linux-headers-5.8.0-7630-generic'
  CC [M]  /home/elliot/projects/me_cleaner/chipsec/drivers/linux/chipsec_km.o
  LD [M]  /home/elliot/projects/me_cleaner/chipsec/drivers/linux/chipsec.o
  MODPOST /home/elliot/projects/me_cleaner/chipsec/drivers/linux/Module.symvers
/home/elliot/projects/me_cleaner/chipsec/drivers/linux/amd64/.cpu.o.cmd: No such file or directory
make[2]: *** [scripts/Makefile.modpost:111: /home/elliot/projects/me_cleaner/chipsec/drivers/linux/Module.symvers] Error 1
make[2]: Target '__modpost' not remade because of errors.
make[1]: *** [Makefile:1693: modules] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.8.0-7630-generic'
make: *** [Makefile:27: chipsec] Error 2
make: Target 'all' not remade because of errors.

npmitche, 2021-01-12:
I was able to get around this issue by adding a command to the make file. See this PR: https://github.com/chipsec/chipsec/pull/1071

Can you let me know if this fixes your issue too?

tinylabs, 2021-01-12:
> I was able to get around this issue by adding a command to the make file. See this PR: #1071
> 
> Can you let me know if this fixes your issue too?

@npmitche 
Yes, that worked! Thank you.

npmitche, 2021-02-27:
@unilock I believe this is fixed. If you are still seeing this issue, please re-open.

closed 2021-02-27 by npmitche