chipsec/chipsec #1040: Chipsec UEFI Fails for version 1.5.6 - ValueError: Empty module name

nstarke, 2020-11-09:
Hi,
I attempted to use chipsec `1.5.6` to dump the SPI ROM off a device and was greeted with this stack trace:

```
FS0:\chipsec\> python -^# chipsec_util.py spi dump rom.bin
Traceback (most recent call last):
  File "chipsec_util.py", line 213, in <module>
    sys.exit( main() )
  File "chipsec_util.py", line 208, in main
    chipsecUtil = ChipsecUtil(argv if argv else sys.argv[1:])
  File "chipsec_util.py", line 71, in __init__
    self.import_cmds()
  File "chipsec_util.py", line 127, in import_cmds
    module = importlib.import_module( cmd_path )
  File "\Efi\StdLib\lib\python.27/importlib/__init__.py", line 37, in import_module
    __import__(name)
ValueError: Empty module name
```

I then attempted to use chipsec `1.5.1` and was able to successfully complete the operation.  I am not sure if this is the latest version that works or not.

frinzell, 2020-11-09:
I'm not able to see that error.  Are you using the most recent Python EFI Shell interpreter?
https://github.com/chipsec/chipsec/tree/master/__install__/UEFI

nstarke, 2020-11-09:
I tried two different versions of the python StdLib/Tools directories in `chipsec_uefi_x64.zip` - the version at the git HEAD and the version bundled with `1.5.6`.  Those were the two failures.  The success came with the version bundled with `1.5.1`.   Its possible this is platform specific, I suppose.

frinzell, 2020-11-09:
Can I get a log file using the -d switch?
We can move this to email if it makes it easier, [chipsec@intel.com](chipsec@intel.com)

nstarke, 2020-11-09:
It's strange; I tried using the `-v` and `-d` switches and I was not able to get any log file written or differing output.  Perhaps since it seems to work on your end we can close this as not reproducible, since I can't offer any additional information?

npmitche, 2020-11-09:
I'm shooting in the dark, but perhaps try to load the latest python.efi files from 1.5.6 from the __install__ directory? We haven't seen this issue what the old version of python, but it would be good to confirm you're using the latest with 1.5.6.

nstarke, 2020-11-10:
I believe I have tried this, I used the version of `python.efi` bundled with the `1.5.6` version release and that was one of the failure cases.  

I'm sorry to report that I don't have any additional information to provide.  Would it be acceptable to you for me to close this issue as not reproducible?

frinzell, 2020-11-10:
Maybe I misread your comment, you need all files in StdLib and Tools.  All within the EFI folder.  Beyond that, it's possible there was some corruption.  Is the error isolated to just that util command?

frinzell, 2021-02-22:
Closing as not reproducible.  Please reopen if needed.  Thanks!

closed 2021-02-22 by frinzell

nstarke, 2021-02-22:
This was almost surely caused by an incomplete filesystem copy; sorry I did not close it sooner.