chipsec/chipsec #1053: ERROR: Message: "Cannot find chipsec.ko module"

startergo, 2020-12-02:
```
root@ubuntu:/home/g5/chipsec# chipsec_util uefi nvram nvar MP51_007F_03B_LOCKED.bin 

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version : 1.5.7
[CHIPSEC] OS      : Linux 5.4.0-56-generic #62-Ubuntu SMP Mon Nov 23 19:20:19 UTC 2020 x86_64
[CHIPSEC] Python  : 3.8.5 (64-bit)

ERROR: Message: "Cannot find chipsec.ko module"
```

frinzell, 2020-12-02:
Hello,

Try building with ‘python setup.py build_ext -i’.

https://github.com/chipsec/chipsec/wiki/Installing-CHIPSEC-in-Linux

Hope this helps.

thanks,

Aaron

startergo, 2020-12-02:
> Try building with ‘python setup.py build_ext -i’.

```
root@ubuntu:/home/g5/chipsec# python setup.py build_ext -i
running build_ext
building the linux driver
creating /home/g5/chipsec/build/lib.linux-x86_64-2.7
creating /home/g5/chipsec/build/lib.linux-x86_64-2.7/drivers
creating /home/g5/chipsec/build/lib.linux-x86_64-2.7/drivers/linux
creating /home/g5/chipsec/build/lib.linux-x86_64-2.7/drivers/linux/amd64
copying drivers/linux/amd64/cpu.o -> /home/g5/chipsec/build/lib.linux-x86_64-2.7/drivers/linux/amd64
copying drivers/linux/amd64/cpu.asm -> /home/g5/chipsec/build/lib.linux-x86_64-2.7/drivers/linux/amd64
copying drivers/linux/AUTHORS -> /home/g5/chipsec/build/lib.linux-x86_64-2.7/drivers/linux
copying drivers/linux/chipsec.mod.c -> /home/g5/chipsec/build/lib.linux-x86_64-2.7/drivers/linux
copying drivers/linux/chipsec_km.c -> /home/g5/chipsec/build/lib.linux-x86_64-2.7/drivers/linux
copying drivers/linux/.chipsec.o.cmd -> /home/g5/chipsec/build/lib.linux-x86_64-2.7/drivers/linux
copying drivers/linux/.chipsec.ko.cmd -> /home/g5/chipsec/build/lib.linux-x86_64-2.7/drivers/linux
copying drivers/linux/Makefile -> /home/g5/chipsec/build/lib.linux-x86_64-2.7/drivers/linux
copying drivers/linux/chipsec.ko -> /home/g5/chipsec/build/lib.linux-x86_64-2.7/drivers/linux
creating /home/g5/chipsec/build/lib.linux-x86_64-2.7/drivers/linux/include
copying drivers/linux/include/chipsec.h -> /home/g5/chipsec/build/lib.linux-x86_64-2.7/drivers/linux/include
copying drivers/linux/COPYING -> /home/g5/chipsec/build/lib.linux-x86_64-2.7/drivers/linux
copying drivers/linux/chipsec.mod -> /home/g5/chipsec/build/lib.linux-x86_64-2.7/drivers/linux
copying drivers/linux/WARNING.txt -> /home/g5/chipsec/build/lib.linux-x86_64-2.7/drivers/linux
copying drivers/linux/modules.order -> /home/g5/chipsec/build/lib.linux-x86_64-2.7/drivers/linux
copying drivers/linux/chipsec.mod.o -> /home/g5/chipsec/build/lib.linux-x86_64-2.7/drivers/linux
copying drivers/linux/chipsec.o -> /home/g5/chipsec/build/lib.linux-x86_64-2.7/drivers/linux
copying drivers/linux/.chipsec.mod.cmd -> /home/g5/chipsec/build/lib.linux-x86_64-2.7/drivers/linux
copying drivers/linux/Module.symvers -> /home/g5/chipsec/build/lib.linux-x86_64-2.7/drivers/linux
copying drivers/linux/.chipsec_km.o.cmd -> /home/g5/chipsec/build/lib.linux-x86_64-2.7/drivers/linux
copying drivers/linux/.chipsec.mod.o.cmd -> /home/g5/chipsec/build/lib.linux-x86_64-2.7/drivers/linux
copying drivers/linux/README -> /home/g5/chipsec/build/lib.linux-x86_64-2.7/drivers/linux
copying drivers/linux/chipsec_km.o -> /home/g5/chipsec/build/lib.linux-x86_64-2.7/drivers/linux
copying drivers/linux/dkms.conf -> /home/g5/chipsec/build/lib.linux-x86_64-2.7/drivers/linux
creating /home/g5/chipsec/build/lib.linux-x86_64-2.7/drivers/linux/i386
copying drivers/linux/i386/cpu.asm -> /home/g5/chipsec/build/lib.linux-x86_64-2.7/drivers/linux/i386
WARNING: could not find /home/g5/chipsec/build/lib.linux-x86_64-2.7/drivers/linux/amd64/.cpu.o.cmd for /home/g5/chipsec/build/lib.linux-x86_64-2.7/drivers/linux/amd64/cpu.o
copying /home/g5/chipsec/build/lib.linux-x86_64-2.7/drivers/linux/chipsec.ko -> chipsec/helper/linux
removing '/home/g5/chipsec/build/lib.linux-x86_64-2.7/drivers' (and everything under it)
building compression executables
creating /home/g5/chipsec/build/lib.linux-x86_64-2.7/chipsec_tools
creating /home/g5/chipsec/build/lib.linux-x86_64-2.7/chipsec_tools/compression
copying chipsec_tools/compression/build.cmd -> /home/g5/chipsec/build/lib.linux-x86_64-2.7/chipsec_tools/compression
creating /home/g5/chipsec/build/lib.linux-x86_64-2.7/chipsec_tools/compression/Brotli
copying chipsec_tools/compression/Brotli/README.md -> /home/g5/chipsec/build/lib.linux-x86_64-2.7/chipsec_tools/compression/Brotli
copying chipsec_tools/compression/Brotli/ReadMe.txt -> /home/g5/chipsec/build/lib.linux-x86_64-2.7/chipsec_tools/compression/Brotli
copying chipsec_tools/compression/Brotli/GNUmakefile -> /home/g5/chipsec/build/lib.linux-x86_64-2.7/chipsec_tools/compression/Brotli
copying chipsec_tools/compression/Brotli/Makefile -> /home/g5/chipsec/build/lib.linux-x86_64-2.7/chipsec_tools/compression/Brotli
creating /home/g5/chipsec/build/lib.linux-x86_64-2.7/chipsec_tools/compression/Brotli/common
copying chipsec_tools/compression/Brotli/common/transform.h -> /home/g5/chipsec/build/lib.linux-x86_64-2.7/chipsec_tools/compression/Brotli/common
copying chipsec_tools/compression/Brotli/common/version.h -> /home/g5/chipsec/build/lib.linux-x86_64-2.7/chipsec_tools/compression/Brotli/common
copying chipsec_tools/compression/Brotli/common/context.h -> /home/g5/chipsec/build/lib.linux-x86_64-2.7/chipsec_tools/compression/Brotli/common
copying chipsec_tools/compression/Brotli/common/dictionary.h -> /home/g5/chipsec/build/lib.linux-x86_64-2.7/chipsec_tools/compression/Brotli/common
copying chipsec_tools/compression/Brotli/common/dictionary.c -> /home/g5/chipsec/build/lib.linux-x86_64-2.7/chipsec_tools/compression/Brotli/common
copying chipsec_tools/compression/Brotli/common/transform.c -> /home/g5/chipsec/build/lib.linux-x86_64-2.7/chipsec_tools/compression/Brotli/common
copying chipsec_tools/compression/Brotli/common/constants.h -> /home/g5/chipsec/build/lib.linux-x86_64-2.7/chipsec_tools/compression/Brotli/common
copying chipsec_tools/compression/Brotli/common/platform.h -> /home/g5/chipsec/build/lib.linux-x86_64-2.7/chipsec_tools/compression/Brotli/common
[... cut ...]
################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version : 1.5.7
[CHIPSEC] OS      : Linux 5.4.0-56-generic #62-Ubuntu SMP Mon Nov 23 19:20:19 UTC 2020 x86_64
[CHIPSEC] Python  : 3.8.5 (64-bit)

ERROR: Message: "Cannot find chipsec.ko module"
```

startergo, 2020-12-02:
Reinstalled it from  another directory. Now it works.

closed 2020-12-02 by startergo