chipsec/chipsec #1054: Add NVRAM GUID FFF12B8D-7696-4C8B-A985-2747075B4F50

startergo, 2020-12-03:
NVRAM volume GUID for Apple cMP is **FFF12B8D-7696-4C8B-A985-2747075B4F50** with header "$VSS":
[image]
Can this be added to 
https://github.com/chipsec/chipsec/blob/f7fe1be07433d8305bcd2b5bb784fef6531c3398/chipsec/hal/uefi_platform.py
MP5,1 NVRAM volume have two $VSS stores.

frinzell, 2020-12-03:
Added this as an enhancement request.  Thanks.

startergo, 2020-12-03:
I found that variable defined here:
https://github.com/chipsec/chipsec/blob/f7fe1be07433d8305bcd2b5bb784fef6531c3398/chipsec/hal/uefi_common.py#L91

frinzell, 2021-01-13:
Closing.  Please reopen if needed.  Thanks!

closed 2021-01-13 by frinzell