chipsec/chipsec #1062: ERROR :4130

jujialin, 2020-12-16:
hi 
anybody meet this issue ?
followed the user guide,and run script ,then  comes this ..

ths!
################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version : 1.5.7
[CHIPSEC] OS      : Windows 10 10.0.18362 AMD64
[CHIPSEC] Python  : 2.7.18 (64-bit)

WARNING: *******************************************************************
WARNING: Chipsec should only be used on test systems!
WARNING: It should not be installed/deployed on production end-user systems.
WARNING: See WARNING.txt
WARNING: *******************************************************************

[CHIPSEC] API mode: using CHIPSEC kernel module API
ERROR: 4130

frinzell, 2020-12-16:
Per #1063, is this error from an AMD platform?

jujialin, 2020-12-17:
> Per #1063, is this error from an AMD platform?

yes ，AMD Ryzen 7 4800U

jujialin, 2020-12-17:
AMD　not  support for now .close this !

closed 2020-12-17 by jujialin