chipsec/chipsec #1064: Does the spi dump function have to depend on the file system?

RockySYR, 2020-12-17:
I tried to encapsulate the spi dump function separately, and encountered many problems.Can you give me some suggestions？ 
 @tweksteen @rountree @hughsie @mtnmts @WilliamLeara

npmitche, 2020-12-17:
Can you elaborate on what you're trying to do and the errors you're seeing?

RockySYR, 2020-12-18:
> Can you elaborate on what you're trying to do and the errors you're seeing?

I wanna dump the entire SPI flash to a file(.bin) . I successfully ran the command python chipsec_util.py spi dump rom.bin. 
Now I want to pack the spi dump related methods into a program file and write to the kernel file but the related function calls about 'chipsec_util spi dump'are too complicated. So can you give me some suggestions？ @npmitche @tweksteen @rountree @hughsie @mtnmts @WilliamLeara

npmitche, 2021-01-13:
The functionality utilizes the framework. It's not simple to pull the functionality out. Feel free to continue the topic under this thread, but I'm going to close the issue as it doesn't directly apply to Chipsec.

closed 2021-01-13 by npmitche