chipsec/chipsec #1086: Can't seem to build in Fedora 33

mandrewsan, 2021-01-24:
Having issues getting this to build and run in Fedora 33.

Initial error I'm seeing after following the build instructions and running `chipsec_util` command:
```
ERROR: Message: "Cannot find chipsec.ko module"
```

Then trying to run `sudo python3 setup.py build_ext -i`, I get the following output:

```
running build_ext
building compression executables
building the linux driver
make: *** [Makefile:20: check_kernel_dir] Error 1
Traceback (most recent call last):
  File "/tmp/chipsec/setup.py", line 251, in <module>
    setup(
  File "/usr/local/lib/python3.9/site-packages/setuptools/__init__.py", line 165, in setup
    return distutils.core.setup(**attrs)
  File "/usr/lib64/python3.9/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/usr/lib64/python3.9/distutils/dist.py", line 966, in run_commands
    self.run_command(cmd)
  File "/usr/lib64/python3.9/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/tmp/chipsec/setup.py", line 183, in run
    driver_build_function()
  File "/tmp/chipsec/setup.py", line 81, in _build_linux_driver
    subprocess.check_output(["make", "-C", build_driver])
  File "/usr/lib64/python3.9/subprocess.py", line 420, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib64/python3.9/subprocess.py", line 524, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['make', '-C', '/tmp/chipsec/build/lib/drivers/linux']' returned non-zero exit status 2.
```

I have already installed the following packages:
```
kernel kernel-devel-$(uname -r) python python-devel gcc nasm redhat-rpm-config elfutils-libelf-devel
```

Any help appreciated!

npmitche, 2021-01-25:
What kernel version are you using? `uname -r`

mandrewsan, 2021-02-01:
5.10.10-200.fc33.x86_64

npmitche, 2021-02-01:
Can you re-test with the latest version (1.5.9)? There were a couple of updates in that release to the Linux driver that should fix the issue.

mandrewsan, 2021-02-02:
Ok, output has changed. Here is what I'm seeing now:

```
running build_ext
building compression executables
building the linux driver
/tmp/chipsec/build/lib/drivers/linux/chipsec_km.c: In function ‘chipsec_lookup_name’:
/tmp/chipsec/build/lib/drivers/linux/chipsec_km.c:1723:10: error: implicit declaration of function ‘get_fs’; did you mean ‘sget_fc’? [-Werror=implicit-function-declaration]
 1723 |  oldfs = get_fs();
      |          ^~~~~~
      |          sget_fc
/tmp/chipsec/build/lib/drivers/linux/chipsec_km.c:1723:10: error: incompatible types when assigning to type ‘mm_segment_t’ from type ‘int’
/tmp/chipsec/build/lib/drivers/linux/chipsec_km.c:1724:2: error: implicit declaration of function ‘set_fs’; did you mean ‘sget_fc’? [-Werror=implicit-function-declaration]
 1724 |  set_fs (KERNEL_DS);
      |  ^~~~~~
      |  sget_fc
/tmp/chipsec/build/lib/drivers/linux/chipsec_km.c:1724:10: error: ‘KERNEL_DS’ undeclared (first use in this function); did you mean ‘KERNFS_NS’?
 1724 |  set_fs (KERNEL_DS);
      |          ^~~~~~~~~
      |          KERNFS_NS
/tmp/chipsec/build/lib/drivers/linux/chipsec_km.c:1724:10: note: each undeclared identifier is reported only once for each function it appears in
/tmp/chipsec/build/lib/drivers/linux/chipsec_km.c:1751:10: error: incompatible types when assigning to type ‘mm_segment_t’ from type ‘int’
 1751 |  oldfs = get_fs();
      |          ^~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:279: /tmp/chipsec/build/lib/drivers/linux/chipsec_km.o] Error 1
make[1]: *** [Makefile:1806: /tmp/chipsec/build/lib/drivers/linux] Error 2
make: *** [Makefile:28: chipsec] Error 2
Traceback (most recent call last):
  File "/tmp/chipsec/setup.py", line 251, in <module>
    setup(
  File "/usr/local/lib/python3.9/site-packages/setuptools/__init__.py", line 165, in setup
    return distutils.core.setup(**attrs)
  File "/usr/lib64/python3.9/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/usr/lib64/python3.9/distutils/dist.py", line 966, in run_commands
    self.run_command(cmd)
  File "/usr/lib64/python3.9/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/tmp/chipsec/setup.py", line 183, in run
    driver_build_function()
  File "/tmp/chipsec/setup.py", line 81, in _build_linux_driver
    subprocess.check_output(["make", "-C", build_driver])
  File "/usr/lib64/python3.9/subprocess.py", line 420, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib64/python3.9/subprocess.py", line 524, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['make', '-C', '/tmp/chipsec/build/lib/drivers/linux']' returned non-zero exit status 2.

```

npmitche, 2021-02-03:
Sadly this isn't as quick of a fix as I was hoping. I'll continue to look into it though and keep you posted. This is a change to the linux kernel that started in 5.10-rc1. For now, if you want to run chipsec, please downgrade your kernel to something before 5.10. 

Reference: http://lkml.iu.edu/hypermail/linux/kernel/2010.3/00552.html

npmitche, 2021-02-05:
@citypw Do you have any thoughts on the best way to fix this?

citypw, 2021-02-06:
@npmitche I send a workaround PR for this issue. It should work since the major GNU/Linux distro ship with KRPOBEs. The lastest version of Fedora seems enabled it by default but I only  tested it on Debian 10 with v5.10.13 and Ubuntu 20.04 with v5.4.

npmitche, 2021-02-08:
@mandrewsan Can you test PR #1095 and confirm it fixes this issue for you?

mandrewsan, 2021-02-08:
The setup script seems to finish successfully. However, when trying to use chipsec_main, I am still getting the error, cannot find chipsec.ko module.

I am running the scripts out of `/tmp`, where I cloned the repo, if it helps.

Here is the output from the setup command, in case you care to see:

```
running build_ext
building compression executables
creating /tmp/chipsec/build
creating /tmp/chipsec/build/lib
creating /tmp/chipsec/build/lib/chipsec_tools
creating /tmp/chipsec/build/lib/chipsec_tools/compression
copying chipsec_tools/compression/get_vsvars.bat -> /tmp/chipsec/build/lib/chipsec_tools/compression
copying chipsec_tools/compression/build.cmd -> /tmp/chipsec/build/lib/chipsec_tools/compression
creating /tmp/chipsec/build/lib/chipsec_tools/compression/Tiano
copying chipsec_tools/compression/Tiano/TianoCompress.h -> /tmp/chipsec/build/lib/chipsec_tools/compression/Tiano
copying chipsec_tools/compression/Tiano/TianoCompress.c -> /tmp/chipsec/build/lib/chipsec_tools/compression/Tiano
copying chipsec_tools/compression/Tiano/Makefile -> /tmp/chipsec/build/lib/chipsec_tools/compression/Tiano
copying chipsec_tools/compression/Tiano/GNUmakefile -> /tmp/chipsec/build/lib/chipsec_tools/compression/Tiano
creating /tmp/chipsec/build/lib/chipsec_tools/compression/Makefiles
copying chipsec_tools/compression/Makefiles/ms.rule -> /tmp/chipsec/build/lib/chipsec_tools/compression/Makefiles
copying chipsec_tools/compression/Makefiles/ms.lib -> /tmp/chipsec/build/lib/chipsec_tools/compression/Makefiles
copying chipsec_tools/compression/Makefiles/ms.common -> /tmp/chipsec/build/lib/chipsec_tools/compression/Makefiles
copying chipsec_tools/compression/Makefiles/ms.app -> /tmp/chipsec/build/lib/chipsec_tools/compression/Makefiles
copying chipsec_tools/compression/Makefiles/lib.makefile -> /tmp/chipsec/build/lib/chipsec_tools/compression/Makefiles
copying chipsec_tools/compression/Makefiles/header.makefile -> /tmp/chipsec/build/lib/chipsec_tools/compression/Makefiles
copying chipsec_tools/compression/Makefiles/footer.makefile -> /tmp/chipsec/build/lib/chipsec_tools/compression/Makefiles
copying chipsec_tools/compression/Makefiles/app.makefile -> /tmp/chipsec/build/lib/chipsec_tools/compression/Makefiles
copying chipsec_tools/compression/Makefiles/NmakeSubdirs.py -> /tmp/chipsec/build/lib/chipsec_tools/compression/Makefiles
copying chipsec_tools/compression/Makefiles/NmakeSubdirs.bat -> /tmp/chipsec/build/lib/chipsec_tools/compression/Makefiles
copying chipsec_tools/compression/Makefile -> /tmp/chipsec/build/lib/chipsec_tools/compression
creating /tmp/chipsec/build/lib/chipsec_tools/compression/LZMA
creating /tmp/chipsec/build/lib/chipsec_tools/compression/LZMA/Sdk
creating /tmp/chipsec/build/lib/chipsec_tools/compression/LZMA/Sdk/DOC
copying chipsec_tools/compression/LZMA/Sdk/DOC/lzma-sdk.txt -> /tmp/chipsec/build/lib/chipsec_tools/compression/LZMA/Sdk/DOC
copying chipsec_tools/compression/LZMA/Sdk/DOC/lzma-history.txt -> /tmp/chipsec/build/lib/chipsec_tools/compression/LZMA/Sdk/DOC
creating /tmp/chipsec/build/lib/chipsec_tools/compression/LZMA/Sdk/C
copying chipsec_tools/compression/LZMA/Sdk/C/Threads.h -> /tmp/chipsec/build/lib/chipsec_tools/compression/LZMA/Sdk/C
copying chipsec_tools/compression/LZMA/Sdk/C/Threads.c -> /tmp/chipsec/build/lib/chipsec_tools/compression/LZMA/Sdk/C
copying chipsec_tools/compression/LZMA/Sdk/C/Precomp.h -> /tmp/chipsec/build/lib/chipsec_tools/compression/LZMA/Sdk/C
copying chipsec_tools/compression/LZMA/Sdk/C/LzmaEnc.h -> /tmp/chipsec/build/lib/chipsec_tools/compression/LZMA/Sdk/C
copying chipsec_tools/compression/LZMA/Sdk/C/LzmaEnc.c -> /tmp/chipsec/build/lib/chipsec_tools/compression/LZMA/Sdk/C
copying chipsec_tools/compression/LZMA/Sdk/C/LzmaDec.h -> /tmp/chipsec/build/lib/chipsec_tools/compression/LZMA/Sdk/C
copying chipsec_tools/compression/LZMA/Sdk/C/LzmaDec.c -> /tmp/chipsec/build/lib/chipsec_tools/compression/LZMA/Sdk/C
copying chipsec_tools/compression/LZMA/Sdk/C/LzHash.h -> /tmp/chipsec/build/lib/chipsec_tools/compression/LZMA/Sdk/C
copying chipsec_tools/compression/LZMA/Sdk/C/LzFindMt.h -> /tmp/chipsec/build/lib/chipsec_tools/compression/LZMA/Sdk/C
copying chipsec_tools/compression/LZMA/Sdk/C/LzFindMt.c -> /tmp/chipsec/build/lib/chipsec_tools/compression/LZMA/Sdk/C
copying chipsec_tools/compression/LZMA/Sdk/C/LzFind.h -> /tmp/chipsec/build/lib/chipsec_tools/compression/LZMA/Sdk/C
copying chipsec_tools/compression/LZMA/Sdk/C/LzFind.c -> /tmp/chipsec/build/lib/chipsec_tools/compression/LZMA/Sdk/C
copying chipsec_tools/compression/LZMA/Sdk/C/CpuArch.h -> /tmp/chipsec/build/lib/chipsec_tools/compression/LZMA/Sdk/C
copying chipsec_tools/compression/LZMA/Sdk/C/Compiler.h -> /tmp/chipsec/build/lib/chipsec_tools/compression/LZMA/Sdk/C
copying chipsec_tools/compression/LZMA/Sdk/C/Bra86.c -> /tmp/chipsec/build/lib/chipsec_tools/compression/LZMA/Sdk/C
copying chipsec_tools/compression/LZMA/Sdk/C/Bra.h -> /tmp/chipsec/build/lib/chipsec_tools/compression/LZMA/Sdk/C
copying chipsec_tools/compression/LZMA/Sdk/C/Alloc.h -> /tmp/chipsec/build/lib/chipsec_tools/compression/LZMA/Sdk/C
copying chipsec_tools/compression/LZMA/Sdk/C/Alloc.c -> /tmp/chipsec/build/lib/chipsec_tools/compression/LZMA/Sdk/C
copying chipsec_tools/compression/LZMA/Sdk/C/7zVersion.h -> /tmp/chipsec/build/lib/chipsec_tools/compression/LZMA/Sdk/C
copying chipsec_tools/compression/LZMA/Sdk/C/7zTypes.h -> /tmp/chipsec/build/lib/chipsec_tools/compression/LZMA/Sdk/C
[... cut ...]
copying drivers/linux/WARNING.txt -> /tmp/chipsec/build/lib/drivers/linux
copying drivers/linux/README -> /tmp/chipsec/build/lib/drivers/linux
copying drivers/linux/Makefile -> /tmp/chipsec/build/lib/drivers/linux
copying drivers/linux/COPYING -> /tmp/chipsec/build/lib/drivers/linux
copying drivers/linux/AUTHORS -> /tmp/chipsec/build/lib/drivers/linux
copying /tmp/chipsec/build/lib/drivers/linux/chipsec.ko -> chipsec/helper/linux
removing '/tmp/chipsec/build/lib/drivers' (and everything under it)
```

npmitche, 2021-02-08:
Hmm, can you confirm `chipsec.ko` is located in `/tmp/chipsec/build/lib/drivers/linux/chipsec.ko` and/or `/tmp/chipsec/chipsec/helper/linux/chipsec.ko`

mandrewsan, 2021-02-09:
It is in neither location. In `/tmp/chipsec/build/lib` I only have a directory called `chipsec_tools`. In `/tmp/chipsec/chipsec/helper/linux` I have the following files:

```
chipsec.ko 
 cpuid.py  
__init__.py  
legacy_pci.py  
linuxhelper.py  
Makefile
```

mandrewsan, 2021-02-09:
Sorry, I was one directory short! This does exist: `/tmp/chipsec/chipsec/helper/linux`

mandrewsan, 2021-02-09:
The manual I was following instructed to call the script a certain way. I modified to calling it like `python3 chipsec_main.py` and it looks like it is working. I only get the missing module error if I call the script directly, like so: `chipsec_main -m chipsec.modules.common.spi_lock`

So I am good to go. Thank you for all of your help with this!

net-wayfarer, 2021-02-17:
Sorry for all the noise coming from Black Arch. Just chiming in that the fix mentioned by [citypw](https://github.com/chipsec/chipsec/issues/1086#issuecomment-774430840) also applies to Black Arch. In due course, I may push a new PR (and a new issue) with regards to something else. For now this patch does appear to resolve the issue.

commit 9a1cead8 (2021-02-17): Fix #1086: Linux kernel v5.10 removed the old good syscalls get_fs/set_fs
in name of security reasons. The current workaround is get the kallsyms_lookup_name()
via kprobe if the kernel >= v5.10.
closed 2021-02-17 by npmitche