chipsec/chipsec #1094: [Linux driver] Use copy_from_kernel_nofault/copy_to_kernel_nofault to access physical memory

fishilico, 2021-02-05:
Hello,
Currently Linux driver uses `memcpy` when reading physical memory: https://github.com/chipsec/chipsec/blob/f27ad9dbf8ab7a0c864605c5fe20666eb8d93ccc/drivers/linux/chipsec_km.c#L446
and uses a direct `copy_from_user` when writing physical memory:
https://github.com/chipsec/chipsec/blob/f27ad9dbf8ab7a0c864605c5fe20666eb8d93ccc/drivers/linux/chipsec_km.c#L513

This causes a kernel panic on some systems when trying to access "forbidden" physical memory, which is quite annoying.

The Linux kernel provides functions `copy_from_kernel_nofault` and `copy_to_kernel_nofault` to perform copy while catching memory faults instead of crashing. These functions are available since Linux 5.8 and were named `probe_kernel_read` and `probe_kernel_write` before commit https://github.com/torvalds/linux/commit/fe557319aa06c23cffc9346000f119547e0f289a. They are also the functions which are used by [`drivers/char/mem.c`](https://github.com/torvalds/linux/blob/v5.10/drivers/char/mem.c) to implement `/dev/mem`.

Would it be acceptable to introduce these "nofault" functions in the Linux driver?

npmitche, 2021-02-26:
I would like to avoid having a spaghetti mess throughout the driver of #if kernel == x, but I think it could be done pretty cleanly in a function-like macro at the top.  I will put this on our list, but if you want to submit a PR, you are more than welcome to.

frinzell, 2021-04-16:
Closing, fix merged.  Please reopen if needed.  Thanks!

fishilico, 2021-04-21:
> Closing, fix merged. Please reopen if needed. Thanks!

https://github.com/chipsec/chipsec/pull/1121 was not about fixing this issue but fixing building with DKMS, which I am using to build the driver. Now I am encountering other issues (for example the driver is not compatible with `CONFIG_CFI_CLANG` yet: it needs to lookup symbols `page_is_ram.cfi_jt` and `phys_mem_access_prot.cfi_jt` instead of the bare ones) but this issue (#1094) still exists.

I do not see a "reopen" button on GitHub. How can I reopen this issue?

frinzell, 2021-04-21:
Reopened.

frinzell, 2021-05-03:
Hi, can you please contact chipsec@intel.com, sounds like there are multiple issues.  Thanks!

closed 2021-12-08 by npmitche