chipsec/chipsec #1100: [Linux DKMS] Fails to find kernel source tree. Patch included in PR.

net-wayfarer, 2021-02-18:
**Environment:**
Distro name and version: Black Arch (rolling release)
Kernel version: 5.10.16-arch1-1 (currently)
Reproducibility Rate: Always.

**Steps to reproduce:**
1. Install the `chipsec` package from Black Arch repository (without the patched Makefile)
2. Witness when `pacman` runs `dkms` via pacman hook and it shows up as failed, along with /path/to/dkms/log file.

**Expected result:**
Kernel module should be built.

**Actual result:**
Please bear in mind this is an old result, without the aforementioned Makefile fix.
```
DKMS make.log for chipsec-1070.7966175 for kernel 5.10.13-arch1-2 (x86_64)
Sun 07 Feb 2021 23:21:53 AEST
make: Entering directory '/var/lib/dkms/chipsec/1070.7966175/build'
Unable to find the Linux source tree.
make: *** [Makefile:20: check_kernel_dir] Error 1
make: Leaving directory '/var/lib/dkms/chipsec/1070.7966175/build'
```

**Severity/Priority**
Major.

**Notes**
* Apologies for bumping and all the noise before on issue #1086. 
* A [workaround](https://github.com/chipsec/chipsec/pull/1099) has been implemented, here was the [original issue with workaround](https://github.com/BlackArch/blackarch/pull/3004)

npmitche, 2021-02-26:
@net-wayfarer Can you confirm this be closed now that #1099 has been merged?

net-wayfarer, 2021-03-04:
Yup. It has been merged. Thank you! I will close this.

closed 2021-03-04 by net-wayfarer