chipsec/chipsec #1115: Wrong CommandPort value when triggering Communicate()-based SMIs

assafcarlsbad, 2021-03-10:
The function `send_smmc_SMI` is used to trigger SMIs which receive their input via the communication buffer. To do so, the function does the following:
  - Writes CommBuffer location and CommBuffer size to the `smmc` structure
  - Writes 0 to I/O ports 0xB3 and 0xB2 to actually trigger the SMI

The value 0 was chosen because that seems to be the value EDK2 uses when given NULL pointers:
```
  IoWrite8 (ICH9_APM_STS, DataPort    == NULL ? 0 : *DataPort);
  IoWrite8 (ICH9_APM_CNT, CommandPort == NULL ? 0 : *CommandPort);
```

However, when I tried running this code on my own laptop (Lenovo ThinkPad T490) the machine would just freeze. After some reverse engineering, I found out that at least in this case, the firmware uses the value 0xFF (and not zero) in order to trigger Communicate()-based SMIs:
[image]

After changing the code to always use 0xFF instead of 0, SMI dispatching seems to work much better:
[image]

To solve this issue in a more generic manner, I suggest adding an extra argument to the `chipsec_util.py smi smmc` command to allow the user to explicitly specify the value of `CommandPort`. To ensure backwards compatibility, this argument can default to 0.

frinzell, 2021-03-12:
Do PR #1084 and #1118 work?

frinzell, 2021-04-06:
Closing, PR #1084 and #1118 merged.  Please reopen if needed.
Thanks!

closed 2021-04-06 by frinzell