chipsec/chipsec #1116: Requesting reclassification or rewording of SPI Flash Descriptor Write Test Failure

MuraliManoharShanm, 2021-03-11:
Hi Team,
We have some systems for which we want to  keep the ability to program the SPI Flash descriptor during BIOS update process. So we would like to keep the  Spi Flash descriptor write access even after shipping from factory while still protecting any malicious write to Spi Flash Descriptor by using the SPI Protected Range Register where one of the range will start from 0 and extend upto the SPI Flash Descriptor end. This enables the write protection at SPI controller level and allows us the flexibility to update  PCH straps and other items in SPI flash descriptor during BIOS update process. 

Since  Chipsec fails the  spi_desc in the above setup, it is a false positive for our customers. Can you reclassify the failure based on the SPI  Protected Range coverage of the flash descriptor or add a note stating the possibility of SPI Protected Range Based protection and other alternative protections.

Here is the failure:
[*] FRAP = 0x00005BDB << SPI Flash Regions Access Permissions Register (SPIBAR + 0x50)
    [00] BRRA             = DB << BIOS Region Read Access 
    [08] BRWA             = 5B << BIOS Region Write Access 
    [16] BMRAG            = 0 << BIOS Master Read Access Grant 
    [24] BMWAG            = 0 << BIOS Master Write Access Grant 
[*] Software access to SPI flash regions: read = 0xDB, write = 0x5B
[-] Software has write access to SPI flash descriptor
[-] FAILED: SPI flash permissions allow SW to write flash descriptor

SPI Protected Ranges
------------------------------------------------------------
PRx (offset) | Value    | Base     | Limit    | WP? | RP?
------------------------------------------------------------
PR0 (74)     | 00007FFF | 07FFF000 | 00000000 | 0   | 0 
PR1 (78)     | 00007FFF | 07FFF000 | 00000000 | 0   | 0 
PR2 (7C)     | 00007FFF | 07FFF000 | 00000000 | 0   | 0 
PR3 (80)     | 00007FFF | 07FFF000 | 00000000 | 0   | 0 
PR4 (84)     | 980F0000 | 00000000 | 0180F000 | 1   | 0 
[+] PASSED: BIOS is write protected

Thanks
Murali Shanmugam
Distinguished Member of Technical Staff 
Dell EMC

frinzell, 2021-03-12:
Hi Murali,

Do you have a suggestion for the note text or a possible change request?

thanks,
Aaron

MuraliManoharShanm, 2021-03-15:
Hi Aaron,
Thanks for looking into this.
Let me work with my team on a draft text. Also, do you want me to open a change request? 
Is that a different ticket. Please let me know how we can do that.
Thanks
Murali

frinzell, 2021-03-16:
Either way works.  If you have a PR, submit it.  Otherwise, text is fine and I can submit the PR.

frinzell, 2021-03-17:
@MuraliManoharShanm, we are looking at getting a release out shortly.  We'll need to get a PR in next week.

MuraliManoharShanm, 2021-03-17:
Hi @frinzell
Thanks for the quick response. 
Here is a proposed text addition (I referenced an existing text added next to fail result for spectre_v2).
[*] running module: chipsec.modules.common.spi_access
.
.
[-] Software has write access to SPI flash descriptor

[-] FAILED: SPI Flash Region Access Permissions are not programmed securely in flash descriptor
[!] System may be using alternative protection by including descriptor region in SPI Protected Range Registers

[*] running module: chipsec.modules.common.spi_desc
.
.
[-] Software has write access to SPI flash descriptor

[-] FAILED: SPI flash permissions allow SW to write flash descriptor
[!] System may be using alternative protection by including descriptor region in SPI Protected Range Registers

Please feel free to modify if needed. This text update should address our immediate concern. In long term we would like to change these to a warning by programmatically checking for the Protected Range Register. 
Thanks
Murali

frinzell, 2021-03-19:
Please see the PR #1130, hopefully this should address any issues.  Thanks

MuraliManoharShanm, 2021-03-24:
Awesome thanks again for the prompt response. Please updated us when the PR is successful and what version the changes will be picked in or if you have any other questions.

Thanks
Murali

frinzell, 2021-04-06:
1.6.1 should include these changes.

MuraliManoharShanm, 2021-04-09:
@frinzell  that's great . Thanks!

frinzell, 2021-04-09:
Closing.  Thanks!

closed 2021-04-09 by frinzell