chipsec/chipsec #1122: [regression] Build module for a kernel not currently running

yaperez-anssi, 2021-03-16:
Hi, commit ff4a27367bbe1324ab1b960c0f9be85beec32f4c introduced in 1.5.10 leads to a regression in our use case.

We are building a bootable USB key with chipsec included in order to test security requirements for hardware acquired by the French administration (https://github.com/ANSSI-FR/chipsec-check). The build process (https://github.com/ANSSI-FR/chipsec-check/blob/master/tools/create-chipsec.sh) uses debootstrap to create a Debian distribution, then checkouts chipsec on it, and then proceeds to build the kernel modules while chrooted inside the Debian distribution.

Previously we used `KERNEL_SRC_DIR` to indicate the path to the kernel sources on the target distribution, but unfortunately  ff4a27367bbe1324ab1b960c0f9be85beec32f4c breaks that behavior because it relies on the *running* kernel information.

A simple fix would be to be able to still be able to override `KSRC`, with something like:
```
diff --git a/drivers/linux/Makefile b/drivers/linux/Makefile
index 2f10d5d..12a3f88 100644
--- a/drivers/linux/Makefile
+++ b/drivers/linux/Makefile
@@ -5,7 +5,7 @@ MODDIR	= $(DESTDIR)/lib/modules
 KVERS	= $(shell uname -r)
 KVER	= $(KVERS)
 VMODDIR = $(MODDIR)/$(KVER)
-KSRC	= $(VMODDIR)/build
+KSRC	?= $(VMODDIR)/build
 
 ifeq (,$(filter %i686 %i386 %i586,$(MACHINE)))
 	elf-size := elf64
```

frinzell, 2021-04-06:
How is this with 1.6.1?

yaperez-anssi, 2021-04-06:
Builds seems to be ok at least (didn't run the built iso but I assume it would have failed noisily if the module hadn't been built).

Thanks for the merge!

closed 2021-04-06 by yaperez-anssi

JohnAZoidberg, 2021-04-09:
> I assume it would have failed noisily if the module hadn't been built

No, I'm having the issue that with your change I can override the kernel source dir again but the module isn't built. See nixos/nixpkgs#117624

JohnAZoidberg, 2021-04-09:
And when I do make it install it with a workaround it's installed at

```
/lib/python3.8/site-packages/drivers/linux/chipsec.ko
```

not

```
/lib/python3.8/site-packages/chipsec/helper/linux/chipsec.ko
```

where it is expecting it.

yaperez-anssi, 2021-04-09:
> > I assume it would have failed noisily if the module hadn't been built
> 
> No, I'm having the issue that with your change I can override the kernel source dir again but the module isn't built. See [NixOS/nixpkgs#117624](https://github.com/NixOS/nixpkgs/pull/117624)

Looking at that MR I assume you're found a fix/workaround in NixOS? I'm unsure if the module path change is really related to the KSRC change anyway.

JohnAZoidberg, 2021-04-09:
Yes I found a workaround, but it's not related to this issue. Should be a separate issue.