chipsec/chipsec #1135: Platform detection fails on VirtualBox

laomaiweng, 2021-03-25:
I'm trying out CHIPSEC by running it in a VirtualBox VM. Platform detection fails because there is no PCI device at `00:00.0`. (There is one however at `00:1f.0` for the PCH.)
As such, the VID/DID reads on `00:00.0` return `0xFFFF`, and `chipsec_main.py` crashes with:

```
[-] Traceback (most recent call last):
  File "/home/user/chipsec/chipsec_main.py", line 481, in main
    self._cs.init( self._platform, self._pch, (not self._no_driver), self._driver_exists, self._to_file, self._from_file )
  File "/home/user/chipsec/chipsec/chipset.py", line 232, in init
    if did in self.chipset_dictionary[vid] and len(self.chipset_dictionary[vid][did]) > 1 and cpuid in self.detection_dictionary.keys():
KeyError: 65535
```

I understand VirtualBox might not be a supported platform but I suspect the raw Python exception is anomalous. Also, it prevents using the `-i` option to ignore platform detection failure.

I've modified the checks in `Chipset.init()` to check for presence of the VID in the chipset dictionary. This works for me, feel free to apply it too if it seems like the right solution!

* https://github.com/chipsec/chipsec/blob/master/chipsec/chipset.py#L232: `if vid in self.chipset_dictionary and did in self.chipset_dictionary[vid] and len(self.chipset_dictionary[vid][did]) > 1 and cpuid in self.detection_dictionary.keys():`
* https://github.com/chipsec/chipsec/blob/master/chipsec/chipset.py#L244: `elif vid in self.chipset_dictionary and did in self.chipset_dictionary[vid]:`

Sae86, 2021-03-29:
I have tried your modifications and they work for me too. Would you like to submit a PR? If not I'd be more than happy to integrate your changes. Thanks!

laomaiweng, 2021-03-29:
PR opened. Thanks for your feedback!

closed 2021-03-29 by laomaiweng