chipsec/chipsec #1159: ord() expected string of length 1, but int found

mayonezo, 2021-04-11:
I get " ERROR: Exception occurred during chipsec.modules.tools.uefi.s3script_modify.run(): 'ord() expected string of length 1, but int found' ". Google tells me this might be a regression because of python3?

Here is the complete console output without all the hex:

```
ubuntu@ubuntu:~/chipsec$ sudo python3 chipsec_main.py -m tools.uefi.s3script_modify -a replace_op mmio_wr 0xFED1F804 0x6009 0x2
################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.6.1
[CHIPSEC] Arguments: -m tools.uefi.s3script_modify -a replace_op mmio_wr 0xFED1F804 0x6009 0x2

****** Chipsec Linux Kernel module is licensed under GPL 2.0
[CHIPSEC] API mode: using CHIPSEC kernel module API
[!]       Unknown PCH: VID = 0x8086, DID = 0x1E55, RID = 0x04; Using Default.
[!]            Results from this system may be incorrect.
[CHIPSEC] OS      : Linux 5.8.0-25-generic #26-Ubuntu SMP Thu Oct 15 10:30:38 UTC 2020 x86_64
[CHIPSEC] Python  : 3.8.6 (64-bit)
[CHIPSEC] Helper  : LinuxHelper (/home/ubuntu/chipsec/chipsec/helper/linux/chipsec.ko)
[CHIPSEC] Platform: Mobile 3rd Generation Core Processor (Ivy Bridge CPU / Panther Point PCH)
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 0154
[CHIPSEC]      RID: 09
[CHIPSEC] PCH     : Default PCH
[CHIPSEC]      VID: FFFF
[CHIPSEC]      DID: FFFF
[CHIPSEC]      RID: FF
 
[+] loaded chipsec.modules.tools.uefi.s3script_modify
[*] running loaded modules ..

[*] running module: chipsec.modules.tools.uefi.s3script_modify
[*] Module arguments (5):
['replace_op', 'mmio_wr', '0xFED1F804', '0x6009', '0x2']
[uefi] searching for EFI variable(s): ['AcpiGlobalVariable']
[uefi] found: AcpiGlobalVariable {af9ffd67-ec10-488a-9dfc-6cbf5ee22c2e} NV+BS+RT variable
[uefi] AcpiGlobalVariable variable data:
00 A0 E9 DA 00 00 00 00                         |         
[uefi] Pointer to ACPI Global Data structure: 0x00000000DAE9A000
[uefi] Decoding ACPI Global Data structure..
[uefi] AcpiVariableSet structure:
00 70 EF DA 00 00 00 00 00 00 05 00 00 00 00 00 |  p              
40 08 0E 00 00 00 00 00 00 70 F4 DA 00 00 00 00 | @        p      
00 80 E3 DA 00 00 00 00 00 A0 F5 DA 00 00 00 00 |                 
00 00 00 DB 00 00 00 00 00 00 00 00 00 00 00 00 |                 
00 00 00 00 00 00 00 00                         |         
[uefi] ACPI Boot-Script table base = 0x00000000DAF47000
[uefi] Found 1 S3 resume boot-scripts
[uefi] S3 resume boot-script at 0x00000000DAF47000
[uefi] Decoding S3 Resume Boot-Script..
S3 Boot Script AA Parser
[uefi] S3 Resume Boot-Script size: 0x3580

[uefi] [++++++++++ S3 Resume Boot-Script Buffer ++++++++++]
...
[x][ =======================================================================
[x][ Module: S3 Resume Boot-Script Testing
[x][ =======================================================================
[*] Looking for 0x2 opcode in the script at 0x00000000DAF47000..
[+] Found opcode at offset 0x2C02
Entry at offset 0x2C02 (len = 0x17, header len = 0x0):
Data:
02 00 17 02 00 00 00 01 00 00 00 04 F8 D1 FE 00 |                 
00 00 00 09 E0 04 00                            |        Decoded:
  Opcode : S3_BOOTSCRIPT_MEM_WRITE (0x0002)
  Width  : 0x02 (4 bytes)
  Address: 0xFED1F804
  Count  : 0x1
  Values : 0x0004E009

[*] Modifying S3 boot script entry at address 0x00000000DAF49C02..
[mem] 0x00000000DAF49C02
[*] Original entry:
ERROR: Exception occurred during chipsec.modules.tools.uefi.s3script_modify.run(): 'ord() expected string of length 1, but int found'

[CHIPSEC] ***************************  SUMMARY  ***************************
[CHIPSEC] Time elapsed            0.183
[CHIPSEC] Modules total           1
[CHIPSEC] Modules failed to run   1:
ERROR: chipsec.modules.tools.uefi.s3script_modify
[CHIPSEC] Modules passed          0:
[CHIPSEC] Modules information     0:
[CHIPSEC] Modules failed          0:
[CHIPSEC] Modules with warnings   0:
[CHIPSEC] Modules not implemented 0:
[CHIPSEC] Modules not applicable  0:
[CHIPSEC] Modules with exceptions 1:
ERROR: chipsec.modules.tools.uefi.s3script_modify
[CHIPSEC] *****************************************************************
```

frinzell, 2021-04-20:
Thanks for finding this.  With the other python related issues, we need a solution that works on both python 3.7 and python 2.7 (for EFI Shell support).

MrMetric, 2021-12-21:
This is a type error caused by Python's dynamic typing. To work around the issue, I made this patch:

    32a33,37
    > def stringtobytes(mstr):
    >     if isinstance(mstr, str):
    >         return mstr.encode("latin_1")
    >     return mstr
    > 
    407,409c412,414
    <     for c in buffer:
    <         num_string += ['{:02X} '.format(ord(c))]
    <         if ( not (c in string.printable) or (c in string.whitespace) ):
    ---
    >     for c in stringtobytes(buffer):
    >         num_string += ['{:02X} '.format(c)]
    >         if ( not (chr(c) in string.printable) or (chr(c) in string.whitespace) ):

I pasted `stringtobytes` rather than importing it from `chipsec.defines` because `chipsec_util` would not run due to a circular dependency error.

ZetaCorvi, 2022-02-03:
I got this error recently, trying bypass bios write protections on X230 with s3script_modify module. I found that error is caused by buffer printing function, so I looked into chipsec/logger.py, and found print_data() function, which seemed more general. I tried to replace print_buffer instances with print_data() wherever they appeared in the script, and it worked!

frinzell, 2022-02-03:
Is this using python 3?

ZetaCorvi, 2022-02-19:
> Is this using python 3?

Yes, latest version

npmitche, 2024-09-18:
Can anyone confirm if this issue is still present? I have not been able to reproduce it. Please re-open if the issue persists.

closed 2024-09-18 by npmitche