chipsec/chipsec #1164: chipsec.modules.common.secureboot.variables:  ''array.array' object has no attribute 'tostring''

dnoliver, 2021-04-19:
Chipsec 1.6.1 fails to execute `python chipsec_main.py -m chipsec.modules.common.secureboot.variables`

```shell
################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.6.1
[CHIPSEC] Arguments: -m chipsec.modules.common.secureboot.variables

****** Chipsec Linux Kernel module is licensed under GPL 2.0
[CHIPSEC] API mode: using CHIPSEC kernel module API
[CHIPSEC] OS      : Linux 5.8.15-301.fc33.x86_64 #1 SMP Thu Oct 15 16:58:06 UTC 2020 x86_64
[CHIPSEC] Python  : 3.9.0 (64-bit)
[CHIPSEC] Helper  : LinuxHelper (/root/chipsec/chipsec/helper/linux/chipsec.ko)
[CHIPSEC] Platform: Mobile 8th Generation Core Processor (Whiskey Lake U 4 Cores)
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 3E34
[CHIPSEC]      RID: 0C
[CHIPSEC] PCH     : Intel 300 series On-Package PCH
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 9D84
[CHIPSEC]      RID: 30

[+] loaded chipsec.modules.common.secureboot.variables
[*] running loaded modules ..

[*] running module: chipsec.modules.common.secureboot.variables
[x][ =======================================================================
[x][ Module: Attributes of Secure Boot EFI Variables
[x][ =======================================================================
ERROR: Exception occurred during chipsec.modules.common.secureboot.variables.run(): ''array.array' object has no attribute 'tostring''

[CHIPSEC] ***************************  SUMMARY  ***************************
[CHIPSEC] Time elapsed            0.002
[CHIPSEC] Modules total           1
[CHIPSEC] Modules failed to run   1:
ERROR: chipsec.modules.common.secureboot.variables
[CHIPSEC] Modules passed          0:
[CHIPSEC] Modules information     0:
[CHIPSEC] Modules failed          0:
[CHIPSEC] Modules with warnings   0:
[CHIPSEC] Modules not implemented 0:
[CHIPSEC] Modules not applicable  0:
[CHIPSEC] Modules with exceptions 1:
ERROR: chipsec.modules.common.secureboot.variables
[CHIPSEC] *****************************************************************
```

The traceback of the exception is:

```shell
ERROR: Exception occurred during chipsec.modules.common.secureboot.variables.run(): ''array.array' object has no attribute 'tostring''
[-] Traceback (most recent call last):
  File "/root/chipsec/chipsec_main.py", line 278, in run_loaded_modules
    result = self.run_module( modx, modx_argv )
  File "/root/chipsec/chipsec_main.py", line 165, in run_module
    raise msg
  File "/root/chipsec/chipsec_main.py", line 159, in run_module
    result = modx.run( module_argv )
  File "/root/chipsec/chipsec/module.py", line 87, in run
    result = self.mod_obj.run(module_argv)
  File "/root/chipsec/chipsec/modules/common/secureboot/variables.py", line 176, in run
    return self.check_secureboot_variable_attributes( do_modify )
  File "/root/chipsec/chipsec/modules/common/secureboot/variables.py", line 101, in check_secureboot_variable_attributes
    sbvars = self._uefi.list_EFI_variables()
  File "/root/chipsec/chipsec/hal/uefi.py", line 483, in list_EFI_variables
    return self.helper.list_EFI_variables()
  File "/root/chipsec/chipsec/helper/oshelper.py", line 393, in list_EFI_variables
    ret = self.helper.list_EFI_variables()
  File "/root/chipsec/chipsec/helper/linux/linuxhelper.py", line 710, in list_EFI_variables
    return self.kern_list_EFI_variables()
  File "/root/chipsec/chipsec/helper/linux/linuxhelper.py", line 811, in kern_list_EFI_variables
    var = self.kern_get_EFI_variable_full(name, guid)
  File "/root/chipsec/chipsec/helper/linux/linuxhelper.py", line 781, in kern_get_EFI_variable_full
    data = buffer[base:base +new_size].tostring()
AttributeError: 'array.array' object has no attribute 'tostring'
```

My secure boot state and vars are:

```shell
[root@localhost chipsec]# mokutil --sb-state
SecureBoot disabled
[root@localhost chipsec]# mokutil --pk
[key 1]
Invalid X509 certificate
[root@localhost chipsec]# mokutil --kek
[key 1]
SHA1 Fingerprint: 31:59:0b:fd:89:c9:d7:4e:d0:87:df:ac:66:33:4b:39:31:25:4b:30
Certificate:
    Data:
        Version: 3 (0x2)
        Serial Number:
            61:0a:d1:88:00:00:00:00:00:03
        Signature Algorithm: sha256WithRSAEncryption
        Issuer: C=US, ST=Washington, L=Redmond, O=Microsoft Corporation, CN=Microsoft Corporation Third Party Marketplace Root
        Validity
            Not Before: Jun 24 20:41:29 2011 GMT
            Not After : Jun 24 20:51:29 2026 GMT
        Subject: C=US, ST=Washington, L=Redmond, O=Microsoft Corporation, CN=Microsoft Corporation KEK CA 2011
        Subject Public Key Info:
            Public Key Algorithm: rsaEncryption
                RSA Public-Key: (2048 bit)
                Modulus:
                    00:c4:e8:b5:8a:bf:ad:57:26:b0:26:c3:ea:e7:fb:
                    57:7a:44:02:5d:07:0d:da:4a:e5:74:2a:e6:b0:0f:
                    ec:6d:eb:ec:7f:b9:e3:5a:63:32:7c:11:17:4f:0e:
                    e3:0b:a7:38:15:93:8e:c6:f5:e0:84:b1:9a:9b:2c:
                    e7:f5:b7:91:d6:09:e1:e2:c0:04:a8:ac:30:1c:df:
                    48:f3:06:50:9a:64:a7:51:7f:c8:85:4f:8f:20:86:
                    ce:fe:2f:e1:9f:ff:82:c0:ed:e9:cd:ce:f4:53:6a:
                    62:3a:0b:43:b9:e2:25:fd:fe:05:f9:d4:c4:14:ab:
                    11:e2:23:89:8d:70:b7:a4:1d:4d:ec:ae:e5:9c:fa:
                    16:c2:d7:c1:cb:d4:e8:c4:2f:e5:99:ee:24:8b:03:
                    ec:8d:f2:8b:ea:c3:4a:fb:43:11:12:0b:7e:b5:47:
                    92:6c:dc:e6:04:89:eb:f5:33:04:eb:10:01:2a:71:
                    e5:f9:83:13:3c:ff:25:09:2f:68:76:46:ff:ba:4f:
                    be:dc:ad:71:2a:58:aa:fb:0e:d2:79:3d:e4:9b:65:
                    3b:cc:29:2a:9f:fc:72:59:a2:eb:ae:92:ef:f6:35:
                    13:80:c6:02:ec:e4:5f:cc:9d:76:cd:ef:63:92:c1:
                    af:79:40:84:79:87:7f:e3:52:a8:e8:9d:7b:07:69:
                    8f:15
                Exponent: 65537 (0x10001)
        X509v3 extensions:
            1.3.6.1.4.1.311.21.1:
                ...
            X509v3 Subject Key Identifier:
                62:FC:43:CD:A0:3E:A4:CB:67:12:D2:5B:D9:55:AC:7B:CC:B6:8A:5F
            1.3.6.1.4.1.311.20.2:
                .
.S.u.b.C.A
            X509v3 Key Usage:
                Digital Signature, Certificate Sign, CRL Sign
            X509v3 Basic Constraints: critical
                CA:TRUE
            X509v3 Authority Key Identifier:
                keyid:45:66:52:43:E1:7E:58:11:BF:D6:4E:9E:23:55:08:3B:3A:22:6A:A8

            X509v3 CRL Distribution Points:

                Full Name:
                  URI:http://crl.microsoft.com/pki/crl/products/MicCorThiParMarRoo_2010-10-05.crl

            Authority Information Access:
                CA Issuers - URI:http://www.microsoft.com/pki/certs/MicCorThiParMarRoo_2010-10-05.crt

    Signature Algorithm: sha256WithRSAEncryption
         d4:84:88:f5:14:94:18:02:ca:2a:3c:fb:2a:92:1c:0c:d7:a0:
         d1:f1:e8:52:66:a8:ee:a2:b5:75:7a:90:00:aa:2d:a4:76:5a:
         ea:79:b7:b9:37:6a:51:7b:10:64:f6:e1:64:f2:02:67:be:f7:
         a8:1b:78:bd:ba:ce:88:58:64:0c:d6:57:c8:19:a3:5f:05:d6:
         db:c6:d0:69:ce:48:4b:32:b7:eb:5d:d2:30:f5:c0:f5:b8:ba:
         78:07:a3:2b:fe:9b:db:34:56:84:ec:82:ca:ae:41:25:70:9c:
         6b:e9:fe:90:0f:d7:96:1f:e5:e7:94:1f:b2:2a:0c:8d:4b:ff:
         28:29:10:7b:f7:d7:7c:a5:d1:76:b9:05:c8:79:ed:0f:90:92:
         9c:c2:fe:df:6f:7e:6c:0f:7b:d4:c1:45:dd:34:51:96:39:0f:
         e5:5e:56:d8:18:05:96:f4:07:a6:42:b3:a0:77:fd:08:19:f2:
         71:56:cc:9f:86:23:a4:87:cb:a6:fd:58:7e:d4:69:67:15:91:
         7e:81:f2:7f:13:e5:0d:8b:8a:3c:87:84:eb:e3:ce:bd:43:e5:
         ad:2d:84:93:8e:6a:2b:5a:7c:44:fa:52:aa:81:c8:2d:1c:bb:
         e0:52:df:00:11:f8:9a:3d:c1:60:b0:e1:33:b5:a3:88:d1:65:
         19:0a:1a:e7:ac:7c:a4:c1:82:87:4e:38:b1:2f:0d:c5:14:87:
         6f:fd:8d:2e:bc:39:b6:e7:e6:c3:e0:e4:cd:27:84:ef:94:42:
         ef:29:8b:90:46:41:3b:81:1b:67:d8:f9:43:59:65:cb:0d:bc:
         fd:00:92:4f:f4:75:3b:a7:a9:24:fc:50:41:40:79:e0:2d:4f:
         0a:6a:27:76:6e:52:ed:96:69:7b:af:0f:f7:87:05:d0:45:c2:
         ad:53:14:81:1f:fb:30:04:aa:37:36:61:da:4a:69:1b:34:d8:
         68:ed:d6:02:cf:6c:94:0c:d3:cf:6c:22:79:ad:b1:f0:bc:03:
         a2:46:60:a9:c4:07:c2:21:82:f1:fd:f2:e8:79:32:60:bf:d8:
         ac:a5:22:14:4b:ca:c1:d8:4b:eb:7d:3f:57:35:b2:e6:4f:75:
         b4:b0:60:03:22:53:ae:91:79:1d:d6:9b:41:1f:15:86:54:70:
         b2:de:0d:35:0f:7c:b0:34:72:ba:97:60:3b:f0:79:eb:a2:b2:
         1c:5d:a2:16:b8:87:c5:e9:1b:f6:b5:97:25:6f:38:9f:e3:91:
         fa:8a:79:98:c3:69:0e:b7:a3:1c:20:05:97:f8:ca:14:ae:00:
         d7:c4:f3:c0:14:10:75:6b:34:a0:1b:b5:99:60:f3:5c:b0:c5:
         57:4e:36:d2:32:84:bf:9e
```

dnoliver, 2021-04-20:
Issue is related to Python3.9. 

With Python 3.9 (default in Fedora 33):

```
[root@localhost chipsec]# python3.9 chipsec_main.py -m common.secureboot.variables
################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.6.1
[CHIPSEC] Arguments: -m common.secureboot.variables

****** Chipsec Linux Kernel module is licensed under GPL 2.0
[CHIPSEC] API mode: using CHIPSEC kernel module API
[CHIPSEC] OS      : Linux 5.11.14-200.fc33.x86_64 #1 SMP Wed Apr 14 15:25:53 UTC 2021 x86_64
[CHIPSEC] Python  : 3.9.2 (64-bit)
[CHIPSEC] Helper  : LinuxHelper (/root/chipsec/chipsec/helper/linux/chipsec.ko)
[CHIPSEC] Platform: Mobile 7th Generation Core Processor (Kabylake H)
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 5910
[CHIPSEC]      RID: 05
[CHIPSEC] PCH     : Intel HM175 (100 series) PCH
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: A152
[CHIPSEC]      RID: 31

[+] loaded chipsec.modules.common.secureboot.variables
[*] running loaded modules ..

[*] running module: chipsec.modules.common.secureboot.variables
[x][ =======================================================================
[x][ Module: Attributes of Secure Boot EFI Variables
[x][ =======================================================================
ERROR: Exception occurred during chipsec.modules.common.secureboot.variables.run(): ''array.array' object has no attribute 'tostring''

[CHIPSEC] ***************************  SUMMARY  ***************************
[CHIPSEC] Time elapsed            0.002
[CHIPSEC] Modules total           1
[CHIPSEC] Modules failed to run   1:
ERROR: chipsec.modules.common.secureboot.variables
[CHIPSEC] Modules passed          0:
[CHIPSEC] Modules information     0:
[CHIPSEC] Modules failed          0:
[CHIPSEC] Modules with warnings   0:
[CHIPSEC] Modules not implemented 0:
[CHIPSEC] Modules not applicable  0:
[CHIPSEC] Modules with exceptions 1:
ERROR: chipsec.modules.common.secureboot.variables
[CHIPSEC] *****************************************************************

```

With Python 3.7:

```
[root@localhost chipsec]# python3.7 chipsec_main.py -m common.secureboot.variables
################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.6.1
[CHIPSEC] Arguments: -m common.secureboot.variables

****** Chipsec Linux Kernel module is licensed under GPL 2.0
[CHIPSEC] API mode: using CHIPSEC kernel module API
[CHIPSEC] OS      : Linux 5.11.14-200.fc33.x86_64 #1 SMP Wed Apr 14 15:25:53 UTC 2021 x86_64
[CHIPSEC] Python  : 3.7.10 (64-bit)
[CHIPSEC] Helper  : LinuxHelper (/root/chipsec/chipsec/helper/linux/chipsec.ko)
[CHIPSEC] Platform: Mobile 7th Generation Core Processor (Kabylake H)
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 5910
[CHIPSEC]      RID: 05
[CHIPSEC] PCH     : Intel HM175 (100 series) PCH
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: A152
[CHIPSEC]      RID: 31

[+] loaded chipsec.modules.common.secureboot.variables
[*] running loaded modules ..

[*] running module: chipsec.modules.common.secureboot.variables
[x][ =======================================================================
[x][ Module: Attributes of Secure Boot EFI Variables
[x][ =======================================================================
[*] Checking protections of UEFI variable 8be4df61-93ca-11d2-aa0d-00e098032b8c:SecureBoot
[*] Checking protections of UEFI variable 8be4df61-93ca-11d2-aa0d-00e098032b8c:SetupMode
[*] Checking protections of UEFI variable 8be4df61-93ca-11d2-aa0d-00e098032b8c:PK
[+] Variable 8be4df61-93ca-11d2-aa0d-00e098032b8c:PK is authenticated (TIME_BASED_AUTHENTICATED_WRITE_ACCESS)
[*] Checking protections of UEFI variable 8be4df61-93ca-11d2-aa0d-00e098032b8c:KEK
[+] Variable 8be4df61-93ca-11d2-aa0d-00e098032b8c:KEK is authenticated (TIME_BASED_AUTHENTICATED_WRITE_ACCESS)
[*] Checking protections of UEFI variable d719b2cb-3d3a-4596-a3bc-dad00e67656f:db
[+] Variable d719b2cb-3d3a-4596-a3bc-dad00e67656f:db is authenticated (TIME_BASED_AUTHENTICATED_WRITE_ACCESS)
[*] Checking protections of UEFI variable d719b2cb-3d3a-4596-a3bc-dad00e67656f:dbx
[+] Variable d719b2cb-3d3a-4596-a3bc-dad00e67656f:dbx is authenticated (TIME_BASED_AUTHENTICATED_WRITE_ACCESS)

[*] Secure Boot appears to be disabled
[+] PASSED: All Secure Boot UEFI variables are protected

[CHIPSEC] ***************************  SUMMARY  ***************************
[CHIPSEC] Time elapsed            0.044
[CHIPSEC] Modules total           1
[CHIPSEC] Modules failed to run   0:
[CHIPSEC] Modules passed          1:
[+] PASSED: chipsec.modules.common.secureboot.variables
[CHIPSEC] Modules information     0:
[CHIPSEC] Modules failed          0:
[CHIPSEC] Modules with warnings   0:
[CHIPSEC] Modules not implemented 0:
[CHIPSEC] Modules not applicable  0:
[CHIPSEC] *****************************************************************
```

Python 3.9 removed `tostring` method from `array.array`. Now, you should use `tobytes`. See https://docs.python.org/3/whatsnew/3.9.html#removed

frinzell, 2021-04-20:
Thanks for the overview of this issue.  Target Python version is around 3.7 but still need Python 2.7 support for EFI Shell implementation.  Need to find a solution that keeps both Py versions happy.

commit bf2f71e9 (2021-04-30): Change tostring to tobytes in linuxhelper.py

Makes Python 3.9 happy
Fixes #1164