chipsec/chipsec #1173: UEFI decode loops forever with given binary

0xfede7c8, 2021-04-28:
[uefi.zip](https://github.com/chipsec/chipsec/files/6394568/uefi.zip)

Running this command on master branch with the attached file:

```bash
python3 chipsec_util.py --no_driver --skip_config uefi decode uefi.bin
```

Will hang forever on:

```bash
################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version : 1.6.1
[CHIPSEC] OS      : Linux 5.4.0-72-generic #80-Ubuntu SMP Mon Apr 12 17:35:00 UTC 2021 x86_64
[CHIPSEC] Python  : 3.8.5 (64-bit)

[CHIPSEC] Helper  : LinuxHelper (None)
[CHIPSEC] Platform: Unrecognized Platform
[CHIPSEC]      VID: FFFF
[CHIPSEC]      DID: FFFF
[CHIPSEC]      RID: FF
[CHIPSEC] PCH     : Unrecognized PCH
[CHIPSEC]      VID: FFFF
[CHIPSEC]      DID: FFFF
[CHIPSEC]      RID: FF
[CHIPSEC] Executing command 'uefi' with args ['decode', 'uefi.bin']

[CHIPSEC] Parsing EFI volumes from 'uefi.bin'..
```

Please note that cpu usage is 100% for one core and memory usage slowly keeps increasing until exhaustion, suggesting a memory leak.

frinzell, 2021-04-30:
Thanks, we'll take a look.

npmitche, 2021-07-20:
@0xfede7c8  Where did you get this binary from? 

It seems to be running into issues because the FVH (Firmware Volume Headers) do not conform the the EFI_FIRMWARE_VOLUME_HEADER spec definition.

0xfede7c8, 2021-07-20:
Can't remember exactly, right now, sorry. 

Probably was extracted from somewhere else which may have break it.
 
Will try to track it back to the source to confirm it. Will update.

npmitche, 2021-07-20:
Thanks, we probably shouldn't go into an infinite loop, but UEFITool can't parse it either: FfsParser::parse: not a single Volume Top File is found, the image may be corrupted

npmitche, 2022-11-30:
Fix is in place (https://github.com/chipsec/chipsec/pull/1607) using the suggestion from #1572.

npmitche, 2022-12-07:
Merged in #1607

closed 2022-12-07 by npmitche