chipsec/chipsec #1193: Two modules that are passed in the older version(1.5.8) but failed(not able to run) in the latest(1.6.2)

Giridhar547, 2021-05-24:
1.5.8:

running module: chipsec.modules.common.bios_ts
[*] BiosInterfaceLockDown (BILD) control = 1
[*] BIOS Top Swap mode is disabled (TSS = 0)
[*] RTC TopSwap control (TS) = 0
[+] PASSED: BIOS Interface is locked (including Top Swap Mode)

[*] running module: chipsec.modules.common.debugenabled
[x][ =======================================================================
[x][ Module: Debug features test
[x][ =======================================================================
..
[+] CPU debug interface state is correct.

[*] Checking DCI register status
[+] DCI Debug is disabled

[*] Module Result
[+] PASSED: All checks have successfully passed

1.6.2:
[CHIPSEC] Modules with exceptions 2:                                      
  ERROR: chipsec.modules.common.bios_ts
  ERROR: chipsec.modules.common.debugenabled    

[*] running module: chipsec.modules.common.bios_ts
[x][ =======================================================================
[x][ Module: BIOS Interface Lock (including Top Swap Mode)
[x][ =======================================================================
[*] BiosInterfaceLockDown (BILD) control = 1
[*] BIOS Top Swap mode is disabled (TSS = 0)
ERROR: Exception occurred during chipsec.modules.common.bios_ts.run(): ''

[*] Checking DCI register status
ERROR: Exception occurred during chipsec.modules.common.debugenabled.run(): ''

Giridhar547, 2021-05-25:
Giving output from verbose mode:

[mmio] SBREGBAR: 0x0000000000000000 (size = 0x1000000)
[-] Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/chipsec-1.6.2-py3.6.egg/chipsec_main.py", line 159, in run_module
    result = modx.run( module_argv )
  File "/usr/local/lib/python3.6/site-packages/chipsec-1.6.2-py3.6.egg/chipsec/module.py", line 87, in run
    result = self.mod_obj.run(module_argv)
  File "/usr/local/lib/python3.6/site-packages/chipsec-1.6.2-py3.6.egg/chipsec/modules/common/bios_ts.py", line 71, in run
    return self.check_bios_iface_lock()
  File "/usr/local/lib/python3.6/site-packages/chipsec-1.6.2-py3.6.egg/chipsec/modules/common/bios_ts.py", line 55, in check_bios_iface_lock
    ts  = self.cs.get_control( 'TopSwap' )
  File "/usr/local/lib/python3.6/site-packages/chipsec-1.6.2-py3.6.egg/chipsec/chipset.py", line 1098, in get_control
    reg_data = self.read_register(reg, cpu_thread)
  File "/usr/local/lib/python3.6/site-packages/chipsec-1.6.2-py3.6.egg/chipsec/chipset.py", line 853, in read_register
    reg_value = self.msgbus.mm_msgbus_reg_read(int(reg['port'], 16), int(reg['offset'], 16))
  File "/usr/local/lib/python3.6/site-packages/chipsec-1.6.2-py3.6.egg/chipsec/hal/msgbus.py", line 183, in mm_msgbus_reg_read
    mmio_addr = self.cs.mmio.get_MMIO_BAR_base_address('SBREGBAR')[0]
  File "/usr/local/lib/python3.6/site-packages/chipsec-1.6.2-py3.6.egg/chipsec/hal/mmio.py", line 217, in get_MMIO_BAR_base_address
    raise Exception
Exception

ERROR: Exception occurred during chipsec.modules.common.bios_ts.run(): ''
[-] Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/chipsec-1.6.2-py3.6.egg/chipsec_main.py", line 278, in run_loaded_modules
    result = self.run_module( modx, modx_argv )
  File "/usr/local/lib/python3.6/site-packages/chipsec-1.6.2-py3.6.egg/chipsec_main.py", line 165, in run_module
    raise msg
  File "/usr/local/lib/python3.6/site-packages/chipsec-1.6.2-py3.6.egg/chipsec_main.py", line 159, in run_module
    result = modx.run( module_argv )
  File "/usr/local/lib/python3.6/site-packages/chipsec-1.6.2-py3.6.egg/chipsec/module.py", line 87, in run
    result = self.mod_obj.run(module_argv)
  File "/usr/local/lib/python3.6/site-packages/chipsec-1.6.2-py3.6.egg/chipsec/modules/common/bios_ts.py", line 71, in run
    return self.check_bios_iface_lock()
  File "/usr/local/lib/python3.6/site-packages/chipsec-1.6.2-py3.6.egg/chipsec/modules/common/bios_ts.py", line 55, in check_bios_iface_lock
    ts  = self.cs.get_control( 'TopSwap' )
  File "/usr/local/lib/python3.6/site-packages/chipsec-1.6.2-py3.6.egg/chipsec/chipset.py", line 1098, in get_control
    reg_data = self.read_register(reg, cpu_thread)
  File "/usr/local/lib/python3.6/site-packages/chipsec-1.6.2-py3.6.egg/chipsec/chipset.py", line 853, in read_register
    reg_value = self.msgbus.mm_msgbus_reg_read(int(reg['port'], 16), int(reg['offset'], 16))
  File "/usr/local/lib/python3.6/site-packages/chipsec-1.6.2-py3.6.egg/chipsec/hal/msgbus.py", line 183, in mm_msgbus_reg_read
    mmio_addr = self.cs.mmio.get_MMIO_BAR_base_address('SBREGBAR')[0]
  File "/usr/local/lib/python3.6/site-packages/chipsec-1.6.2-py3.6.egg/chipsec/hal/mmio.py", line 217, in get_MMIO_BAR_base_address
    raise Exception
Exception

Sae86, 2021-05-25:
Hello @Giridhar547, what platform are you using?

Giridhar547, 2021-05-25:
@Sae86  I am using Cooper Lake
Running by providing correct PCH argument as i couldn't find the appropriate cooperlake value for -p option
[CHIPSEC] Arguments: -p skx --pch PCH_C620

ErikBjorge, 2021-05-26:
This should be fixed with PR #1190.

Sae86, 2021-06-15:
@Giridhar547 have you tried with PR #1190?

Giridhar547, 2021-06-21:
@Sae86  
Sorry for the delay , yes with chipsec 1.6.3 the modules are able to run without any issues

[+] PASSED: chipsec.modules.common.bios_kbrd_buffer
[+] PASSED: chipsec.modules.common.bios_smi
**[+] PASSED: chipsec.modules.common.bios_ts**
[+] PASSED: chipsec.modules.common.bios_wp
[+] PASSED: chipsec.modules.common.cpu.spectre_v2
**[+] PASSED: chipsec.modules.common.debugenabled**
[+] PASSED: chipsec.modules.common.ia32cfg
[+] PASSED: chipsec.modules.common.me_mfg_mode
[+] PASSED: chipsec.modules.common.memlock
[+] PASSED: chipsec.modules.common.smrr
[+] PASSED: chipsec.modules.common.spd_wd
[+] PASSED: chipsec.modules.common.spi_desc
[+] PASSED: chipsec.modules.common.spi_fdopss
[+] PASSED: chipsec.modules.common.spi_lock
[+] PASSED: chipsec.modules.common.uefi.access_uefispec
 
Thanks for the fix, please close the issue

Sae86, 2021-06-21:
That's great! Thanks, @Giridhar547! Closing this issue as fixed.

closed 2021-06-21 by Sae86