chipsec/chipsec #1195: v1.6.2 windows driver compile error

AlanZsm, 2021-05-26:
Following instruction to build kernel driver with old method (WDK 7600.16385.1), it will occur error as below:

d:\chipsec\chipsec-master\drivers\win7>build -cZg
path contains nonexistant c:\users\xxx\.dotnet\tools, removing
path contains nonexistant c:\users\xxx\.dotnet\tools, removing
BUILD: Compile and Link for AMD64
BUILD: Start time: Wed May 26 11:24:06 2021
BUILD: Examining d:\chipsec\chipsec-master\drivers\win7 directory for files to compile.
BUILD: Compiling and Linking d:\chipsec\chipsec-master\drivers\win7 directory
Configuring OACR for 'root:amd64fre' - <OACR on>
May 26 11:24:07 oacr.dll  OACR version change (from build '0' to '30122')
Compiling resources - chipsec_hlpr.rc
Assembling - amd64\cpu.asm
Compiling - driver.c
1>errors in directory d:\chipsec\chipsec-master\drivers\win7
1>d:\chipsec\chipsec-master\drivers\win7\driver.c(330) : error C2220: warning treated as error - no 'object' file generated
Linking Executable - sys\amd64\chipsec_hlpr.sys
1>link : error LNK1181: cannot open input file 'd:\chipsec\chipsec-master\drivers\win7\objfre_win7_amd64\amd64\driver.obj'
BUILD: Finish time: Wed May 26 11:24:15 2021
BUILD: Done

    5 files compiled - 3 Warnings - 1 Error
    1 executable built - 1 Error

frinzell, 2021-05-26:
Does [this ](https://chipsec.github.io/Install%20in%20Windows.html#building)command work?

AlanZsm, 2021-05-27:
No, "python setup.py build_ext -i" command not works. Even with VS2017 installed, it will just stop with below output, but no driver generated.
Actually, if using VS2017, I can build driver succssfully by open chipsec_hlpr.vcxproj & build solution.  But I wish WDK also works.
 
-------
######################
# Install to D:\Security\Chipsec\20210526\chipsec-master\chipsec_tools\compression\Lib\Win32
# Install to D:\Security\Chipsec\20210526\chipsec-master\chipsec_tools\compression\Bin
######################
building the windows driver
**********************************************************************
** Visual Studio 2017 Developer Command Prompt v15.9.26
** Copyright (c) 2017 Microsoft Corporation
**********************************************************************
[vcvarsall.bat] Environment initialized for: 'x64'

frinzell, 2021-06-01:
Is there anything in the build log?  
Also, are you running this from a VS command prompt?

AlanZsm, 2021-06-03:
Yes, I also tried running from VS2017 command prompt, but also finishe with output 'building the windows driver', no actual driver generated.
However, I don't care about this "python setup.py build_ext -i" method success or not, because it also using VS2017, while I can build success with VS2017 by open chipsec_hlpr.vcxproj. 
I prefer to build success with WDK7600.

Sae86, 2021-06-15:
Windows 7 has been EOL'd which means we no longer officially support it along with WDK7. You can upgrade to WDK8 or WDK10 and use `msbuild` (32 bit) or `msbuild /p:Platform=x64` (64 bit) from the VS command prompt. Would this work for you or is there something specific about WDK7 that you prefer?

AlanZsm, 2021-06-16:
Understood, thanks!

closed 2021-06-16 by AlanZsm