chipsec/chipsec #1203: CHIPSEC crashes when run as non-root even when specifying -n and -i

wrffrz, 2021-06-02:
Some operations CHIPSEC supports include things that don't require access to the hardware at all, such as decoding UEFI volumes from files.

However, even when being used in this way and specifying -n (--no_driver) and -i (--ignore_platform), it still attempts to identify the platform by calling iopl() and attempting to perform legacy PCI access.

When running CHIPSEC as non-root, this results in crashes:
```
$ python3 chipsec_util.py -n -i uefi decode firmware.bin
################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version : 1.6.2
[CHIPSEC] OS      : Linux 5.11.0-17-generic #18-Ubuntu SMP Thu May 6 20:10:11 UTC 2021 x86_64
[CHIPSEC] Python  : 3.9.5 (64-bit)

[CHIPSEC] API mode: using OS native API (not using CHIPSEC kernel module)
Segmentation fault (core dumped)
$
```

With the -d (--debug) flag, the last thing that happens before the crash is this:
```
[*] Discovering Bus Configuration:
[*] loading device buses..
Segmentation fault (core dumped)
```

Here's the last bit of output from strace:
```
newfstatat(AT_FDCWD, "/sys/bus/pci/devices/0000:00:00.0/config", {st_mode=S_IFREG|0644, st_size=4096, ...}, 0) = 0
openat(AT_FDCWD, "/sys/bus/pci/devices/0000:00:00.0/config", O_RDONLY|O_CLOEXEC) = 3
newfstatat(3, "", {st_mode=S_IFREG|0644, st_size=4096, ...}, AT_EMPTY_PATH) = 0
ioctl(3, TCGETS, 0x7fff3e325f10)        = -1 ENOTTY (Inappropriate ioctl for device)
lseek(3, 0, SEEK_CUR)                   = 0
lseek(3, 0, SEEK_SET)                   = 0
read(3, "\206\200\0o\0\4\20\0\1\0\0\6\20\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 4096) = 64
close(3)                                = 0
newfstatat(AT_FDCWD, "/sys/bus/pci/devices/0000:00:00.1/config", 0x7fff3e325f10, 0) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/y/work/tools/chipsec/chipsec/helper/linux", {st_mode=S_IFDIR|0755, st_size=14, ...}, 0) = 0
newfstatat(AT_FDCWD, "/y/work/tools/chipsec/chipsec/helper/linux/legacy_pci.py", {st_mode=S_IFREG|0644, st_size=4071, ...}, 0) = 0
newfstatat(AT_FDCWD, "/y/work/tools/chipsec/chipsec/helper/linux/legacy_pci.py", {st_mode=S_IFREG|0644, st_size=4071, ...}, 0) = 0
openat(AT_FDCWD, "/y/work/tools/chipsec/chipsec/helper/linux/__pycache__/legacy_pci.cpython-39.pyc", O_RDONLY|O_CLOEXEC) = 3
newfstatat(3, "", {st_mode=S_IFREG|0644, st_size=2458, ...}, AT_EMPTY_PATH) = 0
ioctl(3, TCGETS, 0x7fff3e325180)        = -1 ENOTTY (Inappropriate ioctl for device)
lseek(3, 0, SEEK_CUR)                   = 0
lseek(3, 0, SEEK_CUR)                   = 0
newfstatat(3, "", {st_mode=S_IFREG|0644, st_size=2458, ...}, AT_EMPTY_PATH) = 0
read(3, "a\r\r\n\0\0\0\0\375\316>`\347\17\0\0\343\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 2459) = 2458
read(3, "", 1)                          = 0
close(3)                                = 0
iopl(3)                                 = -1 EPERM (Operation not permitted)
mmap(NULL, 4096, PROT_READ|PROT_WRITE|PROT_EXEC, MAP_SHARED|MAP_ANONYMOUS, -1, 0) = 0x7f450e45a000
mmap(NULL, 4096, PROT_READ|PROT_WRITE|PROT_EXEC, MAP_SHARED|MAP_ANONYMOUS, -1, 0) = 0x7f450d9ca000
--- SIGSEGV {si_signo=SIGSEGV, si_code=SI_KERNEL, si_addr=NULL} ---
+++ killed by SIGSEGV (core dumped) +++
Segmentation fault (core dumped)
```

CHIPSEC shouldn't attempt to probe the hardware when run as non-root and performing operations that don't need hardware access.

BrentHoltsclaw, 2021-06-02:
Please try  `--skip_config` option.

The `-n` option specifies not to load the chipsec driver, however in linux will fall back to native commands.
The `-i` option allows chipsec to ignore throwing an exception if the platform is not recognized (a configuration file is not loaded).

wrffrz, 2021-06-03:
Thanks Brent, I somehow overlooked that option.  Seems to work properly with the --skip_config option instead.

closed 2021-06-03 by wrffrz