chipsec/chipsec #1205: Github Actions tests fail on new version number PR

npmitche, 2021-06-03:
Hey @fishilico 

When reving the version number, the "Build the driver with DKMS" step fails. Can you take a look at it? 

Example: https://github.com/chipsec/chipsec/runs/2733215222

npmitche, 2021-06-03:
Never mind. I figured it out!

closed 2021-06-03 by npmitche

fishilico, 2021-06-06:
Hello, the version in `drivers/linux/dkms.conf` also needs to be bumped in order to make `chipsec` find the driver on Linux. This is due to `defines.get_version()` being used in https://github.com/chipsec/chipsec/blob/fbed11ae18f452127c221ac4ec4ab1ca973abe9e/chipsec/helper/linux/linuxhelper.py#L131-L137

Actually this was an issue that I encountered a few months ago and that I fixed in https://github.com/chipsec/chipsec/pull/1121 for chipsec version 1.5.10. The fact that the continuous integration system now catches such issues before a new release is nice :)