chipsec/chipsec #1238: common.spd_wd leaves SMBus in a busy state (breaks touchpad on laptop)

kerneis-anssi, 2021-08-05:
Running module common.spd_wd has the effect of breaking the Synaptic touchpad on my Lenovo x280. The only way I found to restore it is to reboot the laptop. Chipset and kernel logs below. I'd be happy to provide more debug information if you tell me what could be useful.

```
$ sudo python3 chipsec_main.py -m common.spd_wd
################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.7.0
[CHIPSEC] Arguments: -m common.spd_wd

****** Chipsec Linux Kernel module is licensed under GPL 2.0
[CHIPSEC] API mode: using CHIPSEC kernel module API
[CHIPSEC] OS      : Linux 5.10.0-8-amd64 #1 SMP Debian 5.10.46-3 (2021-07-28) x86_64
[CHIPSEC] Python  : 3.9.2 (64-bit)
[CHIPSEC] Helper  : LinuxHelper (/home/kerneis/Projects/RD/UEFI/Sources/chipsec/chipsec/helper/linux/chipsec.ko)
[CHIPSEC] Platform: Mobile 8th Generation Core Processor (Kabylake U-Quad Core)
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 5914
[CHIPSEC]      RID: 08
[CHIPSEC] PCH     : PCH-U with iHDCP 2.2 Premium
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 9D4E
[CHIPSEC]      RID: 21
 
[+] loaded chipsec.modules.common.spd_wd
[*] running loaded modules ..

[*] running module: chipsec.modules.common.spd_wd
[x][ =======================================================================
[x][ Module: SPD Write Disable
[x][ =======================================================================
[+] PASSED: SPD Write Disable is set

[CHIPSEC] ***************************  SUMMARY  ***************************
[CHIPSEC] Time elapsed            0.005
[CHIPSEC] Modules total           1
[CHIPSEC] Modules failed to run   0:
[CHIPSEC] Modules passed          1:
[+] PASSED: chipsec.modules.common.spd_wd
[CHIPSEC] Modules information     0:
[CHIPSEC] Modules failed          0:
[CHIPSEC] Modules with warnings   0:
[CHIPSEC] Modules not implemented 0:
[CHIPSEC] Modules not applicable  0:
[CHIPSEC] *****************************************************************
$ sudo dmesg | tail -n 7
[ 4469.581144] Chipsec module loaded 
[ 4469.581146] ** This module exposes hardware & memory access, **
[ 4469.581147] ** which can effect the secure operation of      **
[ 4469.581147] ** production systems!! Use for research only!   **
[ 4469.964780] chipsec cleanup_module 1965: Destroying chipsec device
[ 4469.964925] chipsec cleanup_module 1967: exit
[ 4470.459685] rmi4_physical rmi4-00: Failed to read irqs, code=-6
$ sudo rmmod rmi_smbus rmi_core
$ sudo modprobe rmi_core; sudo modprobe rmi_smbus
$ sudo dmesg |tail 
[ 4941.243001] rmi4_smbus 0-002c: registering SMbus-connected sensor
[ 4941.243129] i801_smbus 0000:00:1f.4: SMBus is busy, can't use it!
[ 4941.243146] rmi4_physical rmi4-00: Read PDT entry at 0x00e9 failed, code: -16.
[ 4941.243147] rmi4_physical rmi4-00: RMI initial reset failed! Continuing in spite of this.
[ 4941.243246] i801_smbus 0000:00:1f.4: SMBus is busy, can't use it!
[ 4941.243260] rmi4_physical rmi4-00: Could not read PDT properties from 0x00ef (code -16). Assuming 0x00.
[ 4941.243358] i801_smbus 0000:00:1f.4: SMBus is busy, can't use it!
[ 4941.243372] rmi4_physical rmi4-00: Read PDT entry at 0x00e9 failed, code: -16.
[ 4941.243373] rmi4_physical rmi4-00: IRQ counting failed with code -16.
[ 4941.243376] rmi4_physical: probe of rmi4-00 failed with error -16
```

kerneis-anssi, 2021-08-05:
The following patch works around the issue for me, demonstrating that it comes from `SPD.detect()` (which reads the SMBus). Obviously, it's not a proper fix:

```
diff --git a/chipsec/modules/common/spd_wd.py b/chipsec/modules/common/spd_wd.py
index 1530c4a..03e0ea9 100644
--- a/chipsec/modules/common/spd_wd.py
+++ b/chipsec/modules/common/spd_wd.py
@@ -66,12 +66,11 @@ class spd_wd(BaseModule):
         spd_wd_reg = self.cs.read_register( 'SMBUS_HCFG' )
         spd_wd = self.cs.get_register_field( 'SMBUS_HCFG', spd_wd_reg, 'SPD_WD' )
 
-        dimms = _spd.detect()
-
         if 1 == spd_wd:
             self.logger.log_passed_check( "SPD Write Disable is set" )
             self.res = ModuleResult.PASSED
         else:
+            dimms = _spd.detect()
             if dimms:
                 self.logger.log_failed_check( "SPD Write Disable is not set and SPDs were detected" )
                 self.res = ModuleResult.FAILED
```

kerneis-anssi, 2021-08-05:
Some logs without the above patch and with `--hal` flag.

[fail.log](https://github.com/chipsec/chipsec/files/6938388/fail.log)

frinzell, 2021-09-15:
Change makes sense.  #1266  captures the changes at least until I can take a look at the `_spd.detect()`.

closed 2021-09-28 by npmitche