chipsec/chipsec #1261: Incorrect port IO results when reading from RTC

throwaway-a, 2021-09-13:
```
python chipsec_util.py io write 0x70 4 0x8
python chipsec_util.py io read 0x71 1
```
Always returns 0 instead of the RTC month field (which I confirmed via RWEverything is set correctly). 
See below

```
C:\Users\ta\Downloads\chipsec-1.7.0>python chipsec_util.py io write 0x70 4 0x8

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version : 1.7.0
[CHIPSEC] OS      : Windows 10 10.0.19043 AMD64
[CHIPSEC] Python  : 3.9.6 (64-bit)

WARNING: *******************************************************************
WARNING: Chipsec should only be used on test systems!
WARNING: It should not be installed/deployed on production end-user systems.
WARNING: See WARNING.txt
WARNING: *******************************************************************

[CHIPSEC] API mode: using CHIPSEC kernel module API
[!]       Unknown PCH: VID = 0x8086, DID = 0x8C4E, RID = 0x05; Using Default.
[!]            Results from this system may be incorrect.
[CHIPSEC] Helper  : Win32Helper (\??\C:\Users\ta\Downloads\chipsec-1.7.0\chipsec\helper\win\win7_amd64\chipsec_hlpr.sys)
[CHIPSEC] Platform: Desktop 4th Generation Core Processor (Haswell CPU / Lynx Point PCH
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 0C00
[CHIPSEC]      RID: 06
[CHIPSEC] PCH     : Default PCH
[CHIPSEC]      VID: FFFF
[CHIPSEC]      DID: FFFF
[CHIPSEC]      RID: FF
[CHIPSEC] Executing command 'io' with args ['write', '0x70', '4', '0x8']

[CHIPSEC] OUT 0x0070 <- 0x00000008 (size = 0x04)
[CHIPSEC] (io) time elapsed 0.000

C:\Users\ta\Downloads\chipsec-1.7.0>python chipsec_util.py io read 0x71 1

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version : 1.7.0
[CHIPSEC] OS      : Windows 10 10.0.19043 AMD64
[CHIPSEC] Python  : 3.9.6 (64-bit)

WARNING: *******************************************************************
WARNING: Chipsec should only be used on test systems!
WARNING: It should not be installed/deployed on production end-user systems.
WARNING: See WARNING.txt
WARNING: *******************************************************************

[CHIPSEC] API mode: using CHIPSEC kernel module API
[!]       Unknown PCH: VID = 0x8086, DID = 0x8C4E, RID = 0x05; Using Default.
[!]            Results from this system may be incorrect.
[CHIPSEC] Helper  : Win32Helper (\??\C:\Users\ta\Downloads\chipsec-1.7.0\chipsec\helper\win\win7_amd64\chipsec_hlpr.sys)
[CHIPSEC] Platform: Desktop 4th Generation Core Processor (Haswell CPU / Lynx Point PCH
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 0C00
[CHIPSEC]      RID: 06
[CHIPSEC] PCH     : Default PCH
[CHIPSEC]      VID: FFFF
[CHIPSEC]      DID: FFFF
[CHIPSEC]      RID: FF
[CHIPSEC] Executing command 'io' with args ['read', '0x71', '1']

[CHIPSEC] IN 0x0071 -> 0x00000000 (size = 0x01)
[CHIPSEC] (io) time elapsed 0.000
```

Sae86, 2021-09-16:
Hi @throwaway-a, does it happen with the latest release (1.7.1)? I tried reproducing using 1.7.1 it but it's reading it correctly for me.

throwaway-a, 2021-09-16:
Yes, actually it looks like it's resolved in 1.7.1

closed 2021-09-16 by throwaway-a