chipsec/chipsec #1267: Does UEFI shell chipsec even work anymore, on account of python 2.7 rather than 3 being included?

throwaway-a, 2021-09-15:
I did the steps from https://chipsec.github.io/USB%20with%20UEFI%20Shell.html

When I went to use chipsec_main.py in the UEFI shell, I got argparse errors. When I used chipsec_util.py I got importlib errors. 

Then when I looked closer at that documentation I noticed it referenced "python.27". So I checked the python.efi --version and it was indeed 2.7.2. Elsewhere in the manual it says "CHIPSEC has deprecated support for Python2 since June 2020". So is this broken on account of needing a Python 3 shell app? Or did something else go wrong?

npmitche, 2021-09-15:
See this comment: https://github.com/chipsec/chipsec/issues/1260#issuecomment-919618105

We're working on getting python3 support for shell. We have attempted to maintain backwards compatibility through now because of the shell support. 

Are you getting the argparse and importlib errors on an intel system using a release?

throwaway-a, 2021-09-15:
Yes, v1.7.1

npmitche, 2022-02-08:
Closing per python 3 for shell being included in release 1.8.0

closed 2022-02-08 by npmitche