chipsec/chipsec #1269: "Unknown PCH" help

throwaway-a, 2021-09-15:
```
[!]       Unknown PCH: VID = 0x8086, DID = 0x1E47, RID = 0x04; Using Default.
[!]            Results from this system may be incorrect.
```

0x01E47 -> https://www.intel.com/content/dam/www/public/us/en/documents/specification-updates/7-series-chipset-pch-spec-update.pdf

```
[!]       Unknown PCH: VID = 0x8086, DID = 0x8C4E, RID = 0x05; Using Default.
[!]            Results from this system may be incorrect.
```

0x8C4E -> https://www.intel.com/content/dam/www/public/us/en/documents/datasheets/8-series-chipset-pch-datasheet.pdf

throwaway-a, 2021-11-01:
You don't need to add new XML files for them, you just need to add these IDs to the existing 7 and 8 series chipset files. Because they work fine, I just want the warnings to go away.

kebinFeng, 2022-04-11:
ERROR: Unknown Platform: VID = 0x8086, DID = 0x4601, RID = 0x04
ERROR: Platform is not supported (Unknown Platform: VID = 0x8086, DID = 0x4601, RID = 0x04).

dscott90, 2022-08-04:
Requested DIDs were added in the following PRs: https://github.com/chipsec/chipsec/pull/1502 and https://github.com/chipsec/chipsec/pull/1443

closed 2022-08-04 by dscott90