chipsec/chipsec #1276: <locks> and lockedby attribute are missing from chipsec_cfg.xsd

kerneis-anssi, 2021-10-01:
Is the chipsec_cfg.xsd actually used to validate XML files or just there for documentation purposes? Also I didn't find any example or documentation of the use of `<locks>` except in `template.xml`. It seems relatively straightforward, but is it fine to start using it now? What about lockedby, it doesn't seem to be used anywhere?

BrentHoltsclaw, 2021-10-02:
@npmitche should have a better idea about the chipsec_cfg.xsd.  I am not aware of it being used.  

I'll work with @Sae86 to get some documentation for `<locks>`. It should be functional post the 1.7.2 release.   The locks data will need to be added into a configuration file for the utility command to have any value. It can be used within modules as well, but the locks functionality needs the  `CONSISTENCY_CHECKING` flag within `chipset.py` to be enabled.

BrentHoltsclaw, 2021-10-02:
The `lockedby` functionality has yet to be implemented. The goal is to improve the reporting of the state of a lock bit, by not only showing the state of the lock, but the state of any other registers/fields that the lock controls.

npmitche, 2021-10-02:
chipsec_cfg.xsd is not currently used to validate, though though it may in the future.

npmitche, 2024-05-02:
Added in PR https://github.com/chipsec/chipsec/pull/2119

closed 2024-05-02 by npmitche