chipsec/chipsec #1277: error while building chipsec using setup.py

prakashyadav008, 2021-10-04:
Hello,
I am trying to build chipsec on debian 11 bullseye and getting the error as given below. I have followed the installation guide through this link "https://chipsec.github.io/Install%20in%20Linux.html". 

`creating /home/bond007/Documents/tools debian/chipsec/build/lib/drivers/linux/include
copying drivers/linux/include/chipsec.h -> /home/bond007/Documents/tools debian/chipsec/build/lib/drivers/linux/include
copying drivers/linux/README -> /home/bond007/Documents/tools debian/chipsec/build/lib/drivers/linux
copying drivers/linux/Makefile -> /home/bond007/Documents/tools debian/chipsec/build/lib/drivers/linux
make[3]: *** No rule to make target 'debian/chipsec/build/lib/drivers/linux'.  Stop.
make[2]: *** [/usr/src/linux-headers-5.10.0-8-common/Makefile:348: __build_one_by_one] Error 2
make[1]: *** [/usr/src/linux-headers-5.10.0-8-common/Makefile:185: __sub-make] Error 2
make: *** [Makefile:33: clean] Error 2
Traceback (most recent call last):
  File "/home/bond007/Documents/tools debian/chipsec/setup.py", line 251, in <module>
    setup(
  File "/usr/lib/python3/dist-packages/setuptools/__init__.py", line 153, in setup
    return distutils.core.setup(**attrs)
  File "/usr/lib/python3.9/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/usr/lib/python3.9/distutils/dist.py", line 966, in run_commands
    self.run_command(cmd)
  File "/usr/lib/python3.9/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/home/bond007/Documents/tools debian/chipsec/setup.py", line 183, in run
    driver_build_function()
  File "/home/bond007/Documents/tools debian/chipsec/setup.py", line 81, in _build_linux_driver
    subprocess.check_output(["make", "-C", build_driver])
  File "/usr/lib/python3.9/subprocess.py", line 424, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.9/subprocess.py", line 528, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['make', '-C', '/home/bond007/Documents/tools debian/chipsec/build/lib/drivers/linux']' returned non-zero exit status 2.
`

frinzell, 2021-10-04:
Hi,

If you run the failing command, below, you may find that it can’t find the kernel source code.  Building requires the kernel headers\etc and the versions need to match exactly with the running kernel.

make -C /home/bond007/Documents/tools debian/chipsec/build/lib/drivers/linux

Hope this helps.

Aaron

prakashyadav008, 2021-10-04:
Thanks for the answer. I had some idea that is the problem but I already have the headers installed which match the running kernel version. You can see below output, when i try to install linux kernel headers. Am I missing some information ?

```
bond007@bond007-x550lc:~$ sudo apt-get install linux-headers-$(uname -r)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
linux-headers-5.10.0-8-amd64 is already the newest version (5.10.46-5).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

```

frinzell, 2021-10-04:
I suspect that the headers package is the "newest" and not precisely align with the reported uname version.  Looks like you are trying to get 5.10.0-8 but are getting 5.10.46-5.

prakashyadav008, 2021-10-05:
The kernel release and the kernel version is:
```
bond007@bond007-x550lc:~/Documents/tools debian/chipsec$ uname -r
5.10.0-8-amd64
bond007@bond007-x550lc:~/Documents/tools debian/chipsec$ uname -v
#1 SMP Debian 5.10.46-5 (2021-09-23)
bond007@bond007-x550lc:~/Documents/tools debian/chipsec$ 
```
Since the kernel version is "5.10.46-5" thats why it is installing/has installed the headers for that as u saw in the previous post. is it a compatibility issue with the kernel ? Do i need some other kernel version and kernel headers?

npmitche, 2021-10-05:
What's the result when you run the command `make -C /home/bond007/Documents/tools debian/chipsec/build/lib/drivers/linux`?

prakashyadav008, 2021-10-06:
It works now i tried the command ``` make -C /home/bond007/Documents/tools\ debian/chipsec/build/lib/drivers/linux```, but it didnt work, so i changed the folder name from "tools debian" to "tools_debian" and it worked. So, I think the build script is not able to recognize spaces. 

```
bond007@bond007-x550lc:~/Documents/tools debian/chipsec$ make -C /home/bond007/Documents/tools\ debian/chipsec/build/lib/drivers/linux
make: Entering directory '/home/bond007/Documents/tools debian/chipsec/build/lib/drivers/linux'
make -C /lib/modules/5.10.0-8-amd64/build M=/home/bond007/Documents/tools debian/chipsec/build/lib/drivers/linux clean
make[1]: Entering directory '/usr/src/linux-headers-5.10.0-8-amd64'
make[3]: *** No rule to make target 'debian/chipsec/build/lib/drivers/linux'.  Stop.
make[2]: *** [/usr/src/linux-headers-5.10.0-8-common/Makefile:348: __build_one_by_one] Error 2
make[1]: *** [/usr/src/linux-headers-5.10.0-8-common/Makefile:185: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.10.0-8-amd64'
make: *** [Makefile:33: clean] Error 2
make: Leaving directory '/home/bond007/Documents/tools debian/chipsec/build/lib/drivers/linux'
bond007@bond007-x550lc:~/Documents/tools debian/chipsec$ ls
AUTHORS  chipsec          chipsec-manual.pdf  chipsec_util.py  debian  drivers      LICENSE  README.md  setup.py  windows_requirements.txt
build    chipsec_main.py  chipsec_tools       COPYING          docs    __install__  README   scripts    tests
bond007@bond007-x550lc:~/Documents/tools debian/chipsec$ cd ..
bond007@bond007-x550lc:~/Documents/tools_debian$ cd chipsec
bond007@bond007-x550lc:~/Documents/tools_debian/chipsec$ ls
AUTHORS  chipsec          chipsec-manual.pdf  chipsec_util.py  debian  drivers      LICENSE  README.md  setup.py  windows_requirements.txt
build    chipsec_main.py  chipsec_tools       COPYING          docs    __install__  README   scripts    tests
bond007@bond007-x550lc:~/Documents/tools_debian/chipsec$ python3 setup.py build_ext -i
running build_ext
building compression executables
building the linux driver
copying /home/bond007/Documents/tools_debian/chipsec/build/lib/drivers/linux/chipsec.ko -> chipsec/helper/linux
removing '/home/bond007/Documents/tools_debian/chipsec/build/lib/drivers' (and everything under it)
bond007@bond007-x550lc:~/Documents/tools_debian/chipsec$ 

```

npmitche, 2022-02-08:
This issue is caused by the Linux kernel make file not liking the space:
`make -C "/lib/modules/5.14.0-kali4-amd64/build" M="/home/user/Downloads/chipsec 1.8.1/build/lib/drivers/linux" clean`
`make: Entering directory '/usr/src/linux-headers-5.14.0-kali4-amd64'`
`/usr/src/linux-headers-5.14.0-kali4-common/Makefile:141: *** building multiple external modules is not supported.  Stop.`

I tried a bunch of different escape methods, but none of them worked, so sadly this has to go under will not fix.

closed 2022-02-08 by npmitche