chipsec/chipsec #1281: "chipsec_util.py io list" throws "MMIO BAR (GPIOBASE) base address is 0"

tandasat, 2021-10-10:
As far as I can tell, GPIOBASE does not exist since PCH-100 at least, and this commit broke the "io list" command.
https://github.com/chipsec/chipsec/commit/591b8631124fa327e5690f3f34ffae567f911be6

```
c>python chipsec_util.py io list

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version : 1.7.2
[CHIPSEC] OS      : Windows 10 10.0.22000 AMD64
[CHIPSEC] Python  : 3.8.5 (64-bit)

WARNING: *******************************************************************
WARNING: Chipsec should only be used on test systems!
WARNING: It should not be installed/deployed on production end-user systems.
WARNING: See WARNING.txt
WARNING: *******************************************************************

[CHIPSEC] API mode: using CHIPSEC kernel module API
[CHIPSEC] Helper  : Win32Helper (\??\C:\Users\tanda\Desktop\RnD\GitHub\chipsec\drivers\win7\x64\Debug\chipsec_hlpr.sys)
[CHIPSEC] Platform: (IceLake U 4 Cores)
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 8A12
[CHIPSEC]      RID: 03
[CHIPSEC] PCH     : Intel 495 series PCH-LP Prem-U
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 3482
[CHIPSEC]      RID: 30
[CHIPSEC] Executing command 'io' with args ['list']

--------------------------------------------------------------------------------
 I/O Range    | BAR Register   | Base             | Size     | En? | Description
--------------------------------------------------------------------------------
 ABASE        | ABASE          | 0000000000001800 | 00000080 | 1   | ACPI Base Address
 PMBASE       | ABASE          | 0000000000001800 | 00000080 | 1   | ACPI Base Address
 TCOBASE      | TCOBASE        | 0000000000000400 | 00000020 | 1   | TCO Base Address
Traceback (most recent call last):
  File "chipsec_util.py", line 218, in <module>
    sys.exit( main() )
  File "chipsec_util.py", line 214, in main
    return chipsecUtil.main()
  File "chipsec_util.py", line 189, in main
    comm.run()
  File "C:\Users\tanda\Desktop\RnD\GitHub\chipsec\chipsec\utilcmd\io_cmd.py", line 111, in run
    self.func()
  File "C:\Users\tanda\Desktop\RnD\GitHub\chipsec\chipsec\utilcmd\io_cmd.py", line 73, in io_list
    self._iobar.list_IO_BARs()
  File "C:\Users\tanda\Desktop\RnD\GitHub\chipsec\chipsec\hal\iobar.py", line 148, in list_IO_BARs
    (_base, _size) = self.get_IO_BAR_base_address( _bar_name )
  File "C:\Users\tanda\Desktop\RnD\GitHub\chipsec\chipsec\hal\iobar.py", line 100, in get_IO_BAR_base_address
    raise CSReadError("MMIO BAR ({}) base address is 0".format(bar_name))
chipsec.exceptions.CSReadError: MMIO BAR (GPIOBASE) base address is 0
```

Sae86, 2021-12-10:
Hi @tandasat, thanks for reporting this. Seems like it's fixed in our latest release (1.8.0), could you please try with it and confirm?

tandasat, 2021-12-10:
It works now. Thank you for checking in. Closing this.
```
>python chipsec_util.py io list

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version : 1.8.0
[CHIPSEC] OS      : Windows 10 10.0.22000 AMD64
[CHIPSEC] Python  : 3.8.5 (64-bit)

WARNING: *******************************************************************
WARNING: Chipsec should only be used on test systems!
WARNING: It should not be installed/deployed on production end-user systems.
WARNING: See WARNING.txt
WARNING: *******************************************************************

[CHIPSEC] API mode: using CHIPSEC kernel module API
[CHIPSEC] Helper  : Win32Helper (\??\C:\Users\tanda\Desktop\RnD\GitHub\chipsec\drivers\win7\x64\Debug\chipsec_hlpr.sys)
[CHIPSEC] Platform: (IceLake U 4 Cores)
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 8A12
[CHIPSEC]      RID: 03
[CHIPSEC] PCH     : Intel 495 series PCH-LP Prem-U
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 3482
[CHIPSEC]      RID: 30
[CHIPSEC] Executing command 'io' with args ['list']

--------------------------------------------------------------------------------
 I/O Range    | BAR Register   | Base             | Size     | En? | Description
--------------------------------------------------------------------------------
 ABASE        | ABASE          | 0000000000001800 | 00000080 | 1   | ACPI Base Address
 PMBASE       | ABASE          | 0000000000001800 | 00000080 | 1   | ACPI Base Address
 TCOBASE      | TCOBASE        | 0000000000000400 | 00000020 | 1   | TCO Base Address
 SMBUS_BASE   | 00:1F.4 + 0x20 | 000000000000EFA0 | 00000080 | 1   | SMBus Base Address
[CHIPSEC] (io) time elapsed 0.016
```

closed 2021-12-10 by tandasat