chipsec/chipsec #1296: scan_image.py

skweiler, 2021-11-03:
> ERROR: Exception occurred during chipsec.modules.tools.uefi.scan_image.run():'list index out of range'

frinzell, 2021-11-03:
Hi, is it possible to forward, chipaec@intel.com, or attach a verbose (-vv) log file?

skweiler, 2021-11-03:
Here is the log.

[log.txt](https://github.com/chipsec/chipsec/files/7470494/log.txt)

frinzell, 2021-11-03:
This switch `-a generate,efilist.json` appears to be missing the FW file argument.  Example: `-a generate,efilist.json,fw.bin`.  If the FW file doesn't exist, removing the `,efilist.json` argument should auto generate both json and bin files.

skweiler, 2021-11-03:
I am very new to this.

> ERROR: Exception occurred during chipsec.modules.tools.uefi.scan_image.run(): 'memoryview: length is not a multiple of itemsize'

[log2.txt](https://github.com/chipsec/chipsec/files/7470740/log2.txt)

frinzell, 2021-11-03:
No worries!  Can you try removing the `-i` and `-n` switches as well?  When generating the FW bin, you will want the driver to load.  If it was only parsing the bin, these switches would be appropriate.

skweiler, 2021-11-03:
I think that this worked. However, there are two messages of note? "Error: Can not allocate memory" and "Unable to get correct file size for NextFwFile corrupt header information."

frinzell, 2021-11-03:
If this was after the FW image was captured and during the parsing of the image, it's possible parts were using a proprietary or non-standard compression.

skweiler, 2021-11-03:
A possible malware issue is what brought me to this point. Could this be a sign of malware?

frinzell, 2021-11-03:
That's tough to say. This tool is good for creating a baseline for future compares but if the initial capture is suspect then this tool may not help you.  It's companion module, `scan_blocked.py`, may but it'll only detect what it knows about.

skweiler, 2021-11-04:
I will be mysterious in case there is a chance that those responsible are able to and do read this and say that I disagree with your second sentence. It is certainly is important that there is **A** clean baseline, but there isn't any reason that it has to be first.

I appreciate your help.

closed 2021-11-04 by skweiler