chipsec/chipsec #1318: Don't run Intel-specific modules on AMD

kerneis-anssi, 2021-12-03:


frinzell, 2022-07-28:
We included a lot of updates around this and believe most modules are addressed.  Using 1.8.7, do you see any modules  that should not run on AMD?

kerneis-anssi, 2022-07-29:
No, this issue is fixed I think. Thanks! I'll send specific PR if I find issues in the future.

Les données à caractère personnel recueillies et traitées dans le cadre de cet échange, le sont à seule fin d’exécution d’une relation professionnelle et s’opèrent dans cette seule finalité et pour la durée nécessaire à cette relation. Si vous souhaitez faire usage de vos droits de consultation, de rectification et de suppression de vos données, veuillez contacter ***@***.*** Si vous avez reçu ce message par erreur, nous vous remercions d’en informer l’expéditeur et de détruire le message. The personal data collected and processed during this exchange aims solely at completing a business relationship and is limited to the necessary duration of that relationship. If you wish to use your rights of consultation, rectification and deletion of your data, please contact: ***@***.*** If you have received this message in error, we thank you for informing the sender and destroying the message.

frinzell, 2022-08-18:
Thanks!  Closing issue.  Please open an issue if something arises.

closed 2022-08-18 by frinzell