chipsec/chipsec #1332: The SPI write command always bricks hardware

TakahiroHaruyama, 2021-12-15:
I'm testing the "chipsec_util spi write" command for my bootkit research but the writing always bricks hardware due to the firmware corruption like invalid checksum and firmware volume decompression error. Is the command still experimental?

My hardware is UP Squared (Apollo Lake). Writing the same firmware image with a SPI programmer (SF-100) works. So I guess there is a bug inside the Chipsec spi write code but I couldn't find the cause. 

I would appreciate it if you have any thought regarding this issue. Thanks.

TakahiroHaruyama, 2021-12-15:
BTW, I compared the data acquired by chipsec (spi dump command) and SF-100. The difference in the BIOS region was just a NVRAM variable (MemoryOverwriteRequestControl), so the reading by chipsec worked well on the Apollo Lake SoC. I think the data offset definition in the XML like FADDR and HSFC is correct.

Sae86, 2022-06-14:
@TakahiroHaruyama are you still seeing this issue with latest CHISPEC release? There have been updates to the code and APL configuration.

Sae86, 2024-04-25:
@TakahiroHaruyama Depending on what SPI protections are in place, writing could be affected. Also, chipsec does not have a way to update checksums automatically but you could manually change them before flashing. 
Using a SPI programmer works better because it doesn't care about the protected regions. 
If you're still seeing any issues please open a new issue and try flashing your FW using chipsec and attaching the binary (read using SPI programmer) so we can help confirm whether it's a chipsec issue or your flash protecting itself.

closed 2024-04-25 by Sae86