chipsec/chipsec #1435: VSS decode error

startergo, 2022-04-13:
```
sudo python3 /Users/x299/Downloads/chipsec/chipsec_util.py decode spi.bin vss

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version : 1.8.3
[CHIPSEC] OS      : Darwin 21.4.0 Darwin Kernel Version 21.4.0: Fri Mar 18 00:45:05 PDT 2022; root:xnu-8020.101.4~15/RELEASE_X86_64 x86_64
[CHIPSEC] Python  : 3.10.4 (64-bit)

[CHIPSEC] API mode: using OS native API (not using CHIPSEC kernel module)
[!]       Unknown Platform: VID = 0xFFFF, DID = 0xFFFF, RID = 0xFF; Using Default.
[!]       Unknown PCH: VID = 0xFFFF, DID = 0xFFFF, RID = 0xFF; Using Default.
[!]            Results from this system may be incorrect.
[CHIPSEC] Helper  : OSXHelper (None)
[CHIPSEC] Platform: UnknownPlatform
[CHIPSEC]      VID: FFFF
[CHIPSEC]      DID: FFFF
[CHIPSEC]      RID: FF
[CHIPSEC] PCH     : Default PCH
[CHIPSEC]      VID: FFFF
[CHIPSEC]      DID: FFFF
[CHIPSEC]      RID: FF
[CHIPSEC] Executing command 'decode' with args ['spi.bin', 'vss']

[CHIPSEC] Decoding SPI ROM image from a file 'spi.bin'
[CHIPSEC] Found SPI Flash descriptor at offset 0x0 in the binary 'spi.bin'
Traceback (most recent call last):
  File "/Users/x299/Downloads/chipsec/chipsec_util.py", line 236, in <module>
    sys.exit(main())
  File "/Users/x299/Downloads/chipsec/chipsec_util.py", line 232, in main
    return chipsecUtil.main()
  File "/Users/x299/Downloads/chipsec/chipsec_util.py", line 204, in main
    comm.run()
  File "/Users/x299/Downloads/chipsec/chipsec/utilcmd/decode_cmd.py", line 127, in run
    self.decode_rom()
  File "/Users/x299/Downloads/chipsec/chipsec/utilcmd/decode_cmd.py", line 113, in decode_rom
    parse_spi_flash_descriptor( self.cs, region_data )
  File "/Users/x299/Downloads/chipsec/chipsec/hal/spi_descriptor.py", line 115, in parse_spi_flash_descriptor
    cs.print_register('FLMAP0', flmap0)
  File "/Users/x299/Downloads/chipsec/chipsec/chipset.py", line 1113, in print_register
    reg = self.get_register_def(reg_name)
  File "/Users/x299/Downloads/chipsec/chipsec/chipset.py", line 747, in get_register_def
    reg_def = self.Cfg.REGISTERS[reg_name]
KeyError: 'FLMAP0'
```

startergo, 2022-04-13:
Another python error on a different command:
```
sudo python3 /Users/x299/Downloads/chipsec/chipsec_util.py iommu list        
Password:

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version : 1.8.3
[CHIPSEC] OS      : Darwin 21.4.0 Darwin Kernel Version 21.4.0: Fri Mar 18 00:45:05 PDT 2022; root:xnu-8020.101.4~15/RELEASE_X86_64 x86_64
[CHIPSEC] Python  : 3.10.4 (64-bit)

Executing: /usr/bin/kmutil unload -p /Users/x299/Downloads/chipsec/chipsec/helper/osx/chipsec.kext
Executing: /usr/bin/kmutil load -p /Users/x299/Downloads/chipsec/chipsec/helper/osx/chipsec.kext
[CHIPSEC] API mode: using CHIPSEC kernel module API
[CHIPSEC] Helper  : OSXHelper (/Users/x299/Downloads/chipsec/chipsec/helper/osx/chipsec.kext)
[CHIPSEC] Platform: Intel Xeon Processor E5/E7 v5 (Skylake)
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 2020
[CHIPSEC]      RID: 07
[CHIPSEC] PCH     : Intel X299 (200 series) PCH
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: A2D2
[CHIPSEC]      RID: 00
[CHIPSEC] Executing command 'iommu' with args ['list']

[CHIPSEC] Enumerating supported IOMMU engines..
dict_keys(['GFXVTD', 'VTD'])
[CHIPSEC] (iommu) time elapsed 0.000
Executing: /usr/bin/kmutil unload -p /Users/x299/Downloads/chipsec/chipsec/helper/osx/chipsec.kext
x299@x299s-iMac-Pro chipsec % sudo python3 /Users/x299/Downloads/chipsec/chipsec_util.py iommu status GFXVTD

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version : 1.8.3
[CHIPSEC] OS      : Darwin 21.4.0 Darwin Kernel Version 21.4.0: Fri Mar 18 00:45:05 PDT 2022; root:xnu-8020.101.4~15/RELEASE_X86_64 x86_64
[CHIPSEC] Python  : 3.10.4 (64-bit)

Executing: /usr/bin/kmutil load -p /Users/x299/Downloads/chipsec/chipsec/helper/osx/chipsec.kext
[CHIPSEC] API mode: using CHIPSEC kernel module API
[CHIPSEC] Helper  : OSXHelper (/Users/x299/Downloads/chipsec/chipsec/helper/osx/chipsec.kext)
[CHIPSEC] Platform: Intel Xeon Processor E5/E7 v5 (Skylake)
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 2020
[CHIPSEC]      RID: 07
[CHIPSEC] PCH     : Intel X299 (200 series) PCH
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: A2D2
[CHIPSEC]      RID: 00
[CHIPSEC] Executing command 'iommu' with args ['status', 'GFXVTD']

==================================================================
[iommu] GFXVTD IOMMU Engine Status:
==================================================================
Traceback (most recent call last):
  File "/Users/x299/Downloads/chipsec/chipsec_util.py", line 236, in <module>
    sys.exit(main())
  File "/Users/x299/Downloads/chipsec/chipsec_util.py", line 232, in main
    return chipsecUtil.main()
  File "/Users/x299/Downloads/chipsec/chipsec_util.py", line 204, in main
    comm.run()
  File "/Users/x299/Downloads/chipsec/chipsec/utilcmd/iommu_cmd.py", line 139, in run
    self.func()
  File "/Users/x299/Downloads/chipsec/chipsec/utilcmd/iommu_cmd.py", line 125, in iommu_status
    self.iommu_engine('status')
  File "/Users/x299/Downloads/chipsec/chipsec/utilcmd/iommu_cmd.py", line 116, in iommu_engine
    elif (cmd == 'status' ): _iommu.dump_IOMMU_status( e )
  File "/Users/x299/Downloads/chipsec/chipsec/hal/iommu.py", line 151, in dump_IOMMU_status
    gsts_reg = self.cs.read_register( vtd + '_GSTS' )
  File "/Users/x299/Downloads/chipsec/chipsec/chipset.py", line 821, in read_register
    if self.mmio.get_MMIO_BAR_base_address(reg['bar'], _bus)[0] != 0:
  File "/Users/x299/Downloads/chipsec/chipsec/hal/mmio.py", line 249, in get_MMIO_BAR_base_address
    raise CSReadError('[mmio] Base address was determined to be 0')
chipsec.exceptions.CSReadError: [mmio] Base address was determined to be 0
```

frinzell, 2022-04-14:
Hi @startergo, when decoding did you try adding ``--no_driver``?

startergo, 2022-04-14:
> Hi @startergo, when decoding did you try adding `--no_driver`?

what is the exact command?

```
sudo python3 /Users/x299/Downloads/chipsec/chipsec_util.py decode --no_driver spi.bin vss

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version : 1.8.3
[CHIPSEC] OS      : Darwin 21.4.0 Darwin Kernel Version 21.4.0: Fri Mar 18 00:45:05 PDT 2022; root:xnu-8020.101.4~15/RELEASE_X86_64 x86_64
[CHIPSEC] Python  : 3.10.4 (64-bit)

usage: 
CHIPSEC can parse an image file containing data from the SPI flash (such as the result of chipsec_util spi dump). This can be critical in forensic analysis.

Examples:

chipsec_util decode spi.bin vss

This will create multiple log files, binaries, and directories that correspond to the sections, firmware volumes, files, variables, etc. stored in the SPI flash.

.. note: 

    It may be necessary to try various options for fw_type in order to correctly parse NVRAM variables. Currently, CHIPSEC does not autodetect the correct format. If the nvram directory does not appear and the list of nvram variables is empty, try again with another type.

>>> chipsec_util decode <rom> [fw_type]

For a list of fw types run:

>>> chipsec_util decode types

Examples:

>>> chipsec_util decode spi.bin vss
chipsec_util.py: error: unrecognized arguments: --no_driver

```

frinzell, 2022-04-14:
Try this...  chipsec_util.py --no_driver decide spi.bin vss

startergo, 2022-04-15:
```
sudo python3 /Users/x299/Downloads/chipsec/chipsec_util.py --no_driver  decode spi.bin vss
Password:

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version : 1.8.3
[CHIPSEC] OS      : Darwin 21.4.0 Darwin Kernel Version 21.4.0: Fri Mar 18 00:45:05 PDT 2022; root:xnu-8020.101.4~15/RELEASE_X86_64 x86_64
[CHIPSEC] Python  : 3.10.4 (64-bit)

[CHIPSEC] API mode: using OS native API (not using CHIPSEC kernel module)
[!]       Unknown Platform: VID = 0xFFFF, DID = 0xFFFF, RID = 0xFF; Using Default.
[!]       Unknown PCH: VID = 0xFFFF, DID = 0xFFFF, RID = 0xFF; Using Default.
[!]            Results from this system may be incorrect.
[CHIPSEC] Helper  : OSXHelper (None)
[CHIPSEC] Platform: UnknownPlatform
[CHIPSEC]      VID: FFFF
[CHIPSEC]      DID: FFFF
[CHIPSEC]      RID: FF
[CHIPSEC] PCH     : Default PCH
[CHIPSEC]      VID: FFFF
[CHIPSEC]      DID: FFFF
[CHIPSEC]      RID: FF
[CHIPSEC] Executing command 'decode' with args ['spi.bin', 'vss']

[CHIPSEC] Decoding SPI ROM image from a file 'spi.bin'
[CHIPSEC] Found SPI Flash descriptor at offset 0x0 in the binary 'spi.bin'
Traceback (most recent call last):
  File "/Users/x299/Downloads/chipsec/chipsec_util.py", line 236, in <module>
    sys.exit(main())
  File "/Users/x299/Downloads/chipsec/chipsec_util.py", line 232, in main
    return chipsecUtil.main()
  File "/Users/x299/Downloads/chipsec/chipsec_util.py", line 204, in main
    comm.run()
  File "/Users/x299/Downloads/chipsec/chipsec/utilcmd/decode_cmd.py", line 127, in run
    self.decode_rom()
  File "/Users/x299/Downloads/chipsec/chipsec/utilcmd/decode_cmd.py", line 113, in decode_rom
    parse_spi_flash_descriptor( self.cs, region_data )
  File "/Users/x299/Downloads/chipsec/chipsec/hal/spi_descriptor.py", line 115, in parse_spi_flash_descriptor
    cs.print_register('FLMAP0', flmap0)
  File "/Users/x299/Downloads/chipsec/chipsec/chipset.py", line 1113, in print_register
    reg = self.get_register_def(reg_name)
  File "/Users/x299/Downloads/chipsec/chipsec/chipset.py", line 747, in get_register_def
    reg_def = self.Cfg.REGISTERS[reg_name]
KeyError: 'FLMAP0'
```

BrentHoltsclaw, 2022-04-15:
Suggest trying `chipsec_util.py --no_driver uefi decode spi.bin vss`.  That option assumes the spi image has already been extracted.  The `chipsec_util decode` option tries to obtain the spi.rom image and then process it.

startergo, 2022-04-16:
> Suggest trying `chipsec_util.py --no_driver uefi decode spi.bin vss`. That option assumes the spi image has already been extracted. The `chipsec_util decode` option tries to obtain the spi.rom image and then process it.

```
C:\Windows\System32>python "C:\Users\G5\Downloads\chipsec-1.8.3\chipsec-1.8.3\chipsec_util.py" decode

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version : 1.8.3
[CHIPSEC] OS      : Windows 10 10.0.22593 AMD64
[CHIPSEC] Python  : 3.8.5 (32-bit)

WARNING: Python architecture (32-bit) is different from OS architecture (64-bit)
usage:
CHIPSEC can parse an image file containing data from the SPI flash (such as the result of chipsec_util spi dump). This can be critical in forensic analysis.

Examples:

chipsec_util decode spi.bin vss

This will create multiple log files, binaries, and directories that correspond to the sections, firmware volumes, files, variables, etc. stored in the SPI flash.

.. note:

    It may be necessary to try various options for fw_type in order to correctly parse NVRAM variables. Currently, CHIPSEC does not autodetect the correct format. If the nvram directory does not appear and the list of nvram variables is empty, try again with another type.

>>> chipsec_util decode <rom> [fw_type]

For a list of fw types run:

>>> chipsec_util decode types

Examples:

>>> chipsec_util decode spi.bin vss

positional arguments:
  <rom>       file to decode
  fw_type     firmware type

optional arguments:
  -h, --help  show this help message and exit
```

npmitche, 2022-05-19:
Can you try using the `uefi` command of chipsec_util.py instead of the `decode` command?

```shell
chipsec_util.py --no_driver uefi decode spi.bin vss
```

npmitche, 2022-12-07:
If you would like more help with this, please share your spi.bin file so we can further debug. Closing issue for now. Please re-open if more help is needed.