chipsec/chipsec #1529: Cannot compile

Michal-Szczepaniak, 2022-07-07:
gcc12
python3.10

```
python setup.py build_ext -i
running build_ext
building compression executables
Sdk/C/LzmaEnc.c: In function ‘LzmaEnc_CodeOneMemBlock’:
Sdk/C/LzmaEnc.c:2641:19: error: storing the address of local variable ‘outStream’ in ‘*p.rc.outStream’ [-Werror=dangling-pointer=]
 2641 |   p->rc.outStream = &outStream.vt;
      |   ~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~
Sdk/C/LzmaEnc.c:2624:28: note: ‘outStream’ declared here
 2624 |   CLzmaEnc_SeqOutStreamBuf outStream;
      |                            ^~~~~~~~~
Sdk/C/LzmaEnc.c:2624:28: note: ‘pp’ declared here
cc1: all warnings being treated as errors
make[1]: *** [../Makefiles/footer.makefile:21: Sdk/C/LzmaEnc.o] Error 1
make: *** [GNUmakefile:90: LZMA] Error 2
Traceback (most recent call last):
  File "setup.py", line 289, in <module>
    ext_modules=extra_kw)
  File "/usr/lib/python2.7/site-packages/setuptools/__init__.py", line 145, in setup
    return distutils.core.setup(**attrs)
  File "/usr/lib64/python2.7/distutils/core.py", line 151, in setup
    dist.run_commands()
  File "/usr/lib64/python2.7/distutils/dist.py", line 953, in run_commands
    self.run_command(cmd)
  File "/usr/lib64/python2.7/distutils/dist.py", line 972, in run_command
    cmd_obj.run()
  File "setup.py", line 180, in run
    self._build_linux_compression()
  File "setup.py", line 101, in _build_linux_compression
    subprocess.check_output(["make", "-C", build_elf, "-f", "GNUmakefile"])
  File "/usr/lib64/python2.7/subprocess.py", line 223, in check_output
    raise CalledProcessError(retcode, cmd, output=output)
subprocess.CalledProcessError: Command '['make', '-C', '/home/foidbgen/chipsec/build/lib/chipsec_tools/compression', '-f', 'GNUmakefile']' returned non-zero exit status 2
```

npmitche, 2022-07-08:
Please verify what version of python you are using: `python --version`

Can you also share what distro/version you're using?

Michal-Szczepaniak, 2022-07-08:
Python is 3.10

I'm using openSUSE tumbleweed. Problem was with gcc12

npmitche, 2022-07-08:
Thanks, I'll look into it. 

I asked confirmation because in the log it calls out `/usr/lib64/python2.7/`

Michal-Szczepaniak, 2022-07-08:
well maybe it picked wrong one by mistake but the error was at compile stage

npmitche, 2022-07-08:
Yes, agreed. This looks to be a duplicate of #1473.

Michal-Szczepaniak, 2022-07-09:
correct

closed 2022-12-07 by npmitche