chipsec/chipsec #1536: The manual says Python 2 is deprecated, but still uses Python 2 for the Linux build instructions

hamishmb, 2022-07-14:
Hi there,

I noticed when building this today that your manual says Python 2 is deprecated, but it still uses Python 2 in the build instructions for Linux. I tried using Python 3 (3.8.10) instead during the build, but it failed. I'm using Linux Mint 20.3 (Ubuntu 20.04 LTS base) with Linux 5.15.0-41 if that's useful information.

npmitche, 2022-07-14:
Yeah, we still have some floating python2 references that we're working on cleaning up. Did you try the build commands found here: https://chipsec.github.io/installation/Install%20in%20Linux.html ?

Can you provide logs for the build failure you're seeing?

hamishmb, 2022-07-15:
Hmm, that's interesting, it's working now, both for the latest commit and v1.8.6. Nevermind.

closed 2022-07-15 by hamishmb