chipsec/chipsec #1537: Linux: Build and install instructions are split across multiple places in the user manual

hamishmb, 2022-07-14:
The Linux build and install instructions are split across multiple places in the manual - this is fairly unclear and makes it hard to know how to run the finished build.

In retrospect, it seems obvious a setup.py install is needed, but considering a kernel module is involved, this isn't immediately apparent.

npmitche, 2022-07-14:
Can you reference which sections specifically are confusing?

hamishmb, 2022-07-15:
This section includes the build instructions: [https://chipsec.github.io/installation/Install%20in%20Linux.html](https://chipsec.github.io/installation/Install%20in%20Linux.html)

The install instructions for Linux, however, are instead only here: [https://chipsec.github.io/installation/Using-CHIPSEC-with-Kali-Linux.html?highlight=setup%20py%20install](https://chipsec.github.io/installation/Using-CHIPSEC-with-Kali-Linux.html?highlight=setup%20py%20install).

closed 2022-07-26 by npmitche

hamishmb, 2022-07-26:
Cheers :)