chipsec/chipsec #1541: Oracle Linux: issue gcc: error: unrecognized command-line option ‘-ftrivial-auto-var-init=zero’ in setup

lignay, 2022-07-20:
```
running build_ext
building compression executables
building the linux driver
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (GCC) 11.2.1 20220127 (Red Hat 11.2.1-9.1.0.2)
  You are using:           gcc (GCC) 11.2.0
gcc: error: unrecognized command-line option ‘-ftrivial-auto-var-init=zero’
make[2]: *** [scripts/Makefile.build:277: /root/chipsec/build/lib/drivers/linux/chipsec_km.o] Error 1
make[1]: *** [Makefile:1950: /root/chipsec/build/lib/drivers/linux] Error 2
make: *** [Makefile:30: chipsec] Error 2
Traceback (most recent call last):
  File "/root/chipsec/./setup.py", line 261, in <module>
    setup(name='chipsec', version=version(), description='CHIPSEC: Platform Security Assessment Framework',
  File "/usr/local/lib/python3.10/site-packages/setuptools/__init__.py", line 153, in setup
    return distutils.core.setup(**attrs)
  File "/usr/local/lib/python3.10/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/usr/local/lib/python3.10/distutils/dist.py", line 966, in run_commands
    self.run_command(cmd)
  File "/usr/local/lib/python3.10/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/root/chipsec/./setup.py", line 189, in run
    driver_build_function()
  File "/root/chipsec/./setup.py", line 84, in _build_linux_driver
    subprocess.check_output(["make", "-C", build_driver])
  File "/usr/local/lib/python3.10/subprocess.py", line 420, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/local/lib/python3.10/subprocess.py", line 524, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['make', '-C', '/root/chipsec/build/lib/drivers/linux']' returned non-zero exit status 2.
```

lignay, 2022-12-25:
I could not reproduced this issue because of no test environment, the original environment has redeployed for another task. I will provide the details till I have env and I can reproduce it.

npmitche, 2024-09-18:
Closing as cannot duplicate. Please re-open (or submit a new one) if you are able to reproduce this and provide more information.

closed 2024-09-18 by npmitche