chipsec/chipsec #1542: Unknown PCH DID=0x5182

kerneis-anssi, 2022-07-22:
Trying to use the latest chipsec version (from github) on Alderlake and I get an error about unknown PCH (DID=0x5182 RID=0x1).

kerneis-anssi, 2022-07-26:
Datasheet seems to be this one: https://edc.intel.com/content/www/us/en/design/ipla/software-development-platforms/client/platforms/alder-lake-mobile-p/intel-600-series-chipset-family-on-package-platform-controller-hub-pch-datash/002/introduction/

kerneis-anssi, 2022-07-26:
As a temporary work-around for people trying to dump the SPI, forcing the PCH to PCH_4XX seems to work (but is probably not recommended in the general case, it's just that common.xml has a very restricted version of HSFS).

frinzell, 2022-07-26:
Using the `--pch` switch and specifying the PCH will do this for you.

kerneis-anssi, 2022-07-26:
That's what I meant, using --pch PCH_4XX, but I don't think there currently exists a proper pch XML in the public chipsec repo, hence my wording of "workaround".

Les données à caractère personnel recueillies et traitées dans le cadre de cet échange, le sont à seule fin d’exécution d’une relation professionnelle et s’opèrent dans cette seule finalité et pour la durée nécessaire à cette relation. Si vous souhaitez faire usage de vos droits de consultation, de rectification et de suppression de vos données, veuillez contacter ***@***.*** Si vous avez reçu ce message par erreur, nous vous remercions d’en informer l’expéditeur et de détruire le message. The personal data collected and processed during this exchange aims solely at completing a business relationship and is limited to the necessary duration of that relationship. If you wish to use your rights of consultation, rectification and deletion of your data, please contact: ***@***.*** If you have received this message in error, we thank you for informing the sender and destroying the message.

npmitche, 2023-01-06:
Should be fixed with PR #1557

closed 2023-01-06 by npmitche