chipsec/chipsec #1563: uefi shell not return

another1024, 2022-08-19:
hi,itry to run uefi shell on qemu ovmf,but i get some problem.
when i try to run Python368.efi chipsec_main.py,the shell cant return.
here is command:sudo qemu-system-x86_64 -bios OVMF.fd -hdd hda.img -m 1024
how to find problem? can you help me?

another1024, 2022-08-19:
OVMF is build by  edk2
 build -a X64 -t GCC5 -p OvmfPkg/OvmfPkgX64.dsc  -D SOURCE_DEBUG_ENABLE

frinzell, 2022-08-19:
Can you build the default EDK2-LIBC python368?

fishilico, 2022-08-27:
Hello, I also encountered some issues when I tried using chipsec in a UEFI shell in a QEMU virtual machine, some time ago. I eventually managed to get it working reliably and I just created two PR to make it easier:
- https://github.com/chipsec/chipsec/pull/1571 adds a standalone Python scripts which creates a ESP with `python368.efi` from https://github.com/chipsec/chipsec/tree/main/__install__/UEFI . It uses `-drive if=pflash,format=raw,readonly=on,file=/path/to/OVMF_CODE.fd` to start OVMF (and the path to `OVMF_CODE.fd` can be provided as argument)
- https://github.com/chipsec/chipsec/pull/1570 adds support in chipsec for the platform emulated by the default QEMU virtual machine.

Also, I am not sure that the default unit of `m` option is. I remember having issues because the default amount of memory was too low to load all the Python files that chipsec uses, and it may be what happens with `-m 1024`. Could you try with `-m 1G`? (In my script I used `512M` by default and it worked fine, but using a bigger size makes sure that this is not a memory allocation issue)

closed 2022-09-26 by npmitche