chipsec/chipsec #1565: python368 urllib.request import error - no module named _socket

charge007, 2022-08-22:
Hi

Trying to import urllib.request or socket fails with the following issue :

`
>>> import socket
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "FS0:\Efi\StdLib\lib\python36.8\socket.py", line 49, in <module>
    import _socket
ModuleNotFoundError: No module named '_socket'
`

frinzell, 2022-08-23:
Hi, are you building the Python 368 clean from the EDK2-LIBC repo?

frinzell, 2022-08-23:
Ah, I think I see, you are simply trying to import urllib.request?
This does not appear in the list of [tested library modules ](https://github.com/intel-innersource/applications.validation.advancedsystemtools.products.python-efi/blob/main/Python/Python-3.6.8/Py368ReadMe.txt) (end of doc).  May have to approach the [Python 3.6.8](https://github.com/intel-innersource/applications.validation.advancedsystemtools.products.python-efi/tree/main/Python/Python-3.6.8) code maintainer.  We've seen library issues in the past that were due to an unexpected environment (ie the shell).

charge007, 2022-08-24:
Correct, import socket fails as well.
I tried adding more stuff to sys.path but it doesnt help.
Is there any other way of getting these libs to work in uefi shell ? 
Thanks

frinzell, 2022-08-25:
The [Python 3.6.8](https://github.com/intel-innersource/applications.validation.advancedsystemtools.products.python-efi/tree/main/Python/Python-3.6.8) code maintainer would be in a better position to address this.  Recommend filing this with the [TianoCore Bugzilla](https://bugzilla.tianocore.org/).  For reference: here is the [uuid.py issue](https://bugzilla.tianocore.org/show_bug.cgi?id=3899) that was recently fixed.  Hope this helps.

frinzell, 2023-04-10:
EDK2-LIBC BZ issue created:
https://bugzilla.tianocore.org/show_bug.cgi?id=4406

frinzell, 2023-04-12:
@charge007 to enable socket support, you will need to build the EFI Python using step 6 in from the [Py368ReadMe.txt](https://github.com/tianocore/edk2-libc/blob/master/AppPkg/Applications/Python/Python-3.6.8/Py368ReadMe.txt)

As described in the readme, you will need to enable the commented out code in these files:
https://github.com/tianocore/edk2-libc/blob/master/AppPkg/Applications/Python/Python-3.6.8/PyMod-3.6.8/Modules/config.c
https://github.com/tianocore/edk2-libc/blob/master/AppPkg/Applications/Python/Python-3.6.8/Python368.inf

You will have to build using the MSVS tool chain.

Also, recommend using the latest EDK2-LIBC code, all CHIPSEC added support was ported to it just recently.  No need to merge in any files from this repo.

frinzell, 2023-04-13:
Closing this as not a CHIPSEC tool issue.  The Tianocore BZ ticket will remain open and I'll continue working with the dev team to see what can be done.  Until then, following the build steps described in the Py368ReadMe.txt should give the support needed.  Feel free to reopen or follow up as needed.  Thanks.

closed 2023-04-13 by frinzell

charge007, 2023-04-17:
Thank you very much sir